lnis-uofu/OpenFPGA #10: Compile error in ace2

mithro, 2019-07-16:
```
tansell@tansell-glaptop:~/github/LNIS-Projects/OpenFPGA/build$ make
[  1%] Compile Yosys with given Makefile

  Build successful.

[  1%] Built target yosys
[ 83%] Built target libabc
[ 83%] Built target abc
[ 83%] Building CXX object ace2/CMakeFiles/libace.dir/SRC/io_ace.c.o
In file included from /home/tansell/github/LNIS-Projects/OpenFPGA/abc/src/misc/extra/extra.h:44:0,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/abc/src/base/abc/abc.h:42,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/ace.h:6,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:3:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c: In function ‘void ace_io_print_activity(Abc_Ntk_t*, FILE*)’:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:37:3: error: invalid static_cast from type ‘Abc_Obj_t* {aka Abc_Obj_t_*}’ to type ‘int’
   assert(obj->pCopy);
   ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c: In function ‘int ace_io_read_activity(Abc_Ntk_t*, FILE*, ace_pi_format_t, double, double, const char*)’:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:272:13: error: invalid static_cast from type ‘char*’ to type ‘int’
             assert(res);
             ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:295:17: error: invalid static_cast from type ‘char*’ to type ‘int’
                 assert(res);
                 ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:308:13: error: invalid static_cast from type ‘char*’ to type ‘int’
             assert(res);
             ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:311:17: error: invalid static_cast from type ‘char*’ to type ‘int’
                 assert(res);
                 ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:328:13: error: invalid static_cast from type ‘char*’ to type ‘int’
             assert(res);
             ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:367:17: error: invalid static_cast from type ‘char*’ to type ‘int’
                 assert(res);
                 ^
ace2/CMakeFiles/libace.dir/build.make:127: recipe for target 'ace2/CMakeFiles/libace.dir/SRC/io_ace.c.o' failed
make[2]: *** [ace2/CMakeFiles/libace.dir/SRC/io_ace.c.o] Error 1
CMakeFiles/Makefile2:306: recipe for target 'ace2/CMakeFiles/libace.dir/all' failed
make[1]: *** [ace2/CMakeFiles/libace.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

tangxifan, 2019-07-16:
Hi, would you mind sharing your OS and compiler info, which may help us to reproduce the error?
We have tested on CentOS with gcc-8.2. 
Similarly, you can find the dependency requirements for different OS on our TravisCI configuration file.

Best regards
Xifan Tang

mithro, 2019-07-16:
```
tansell@tansell-glaptop:~/github/LNIS-Projects/OpenFPGA/build$ cmake .. -DCMAKE_BUILD_TYPE=debug
-- CMAKE_BUILD_TYPE: debug
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2 - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcatch-value=3
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcatch-value=3 - Failed
-- CMAKE_CXX_FLAGS:   -Wpointer-arith -Wcast-qual -D__USE_FIXED_PROTOTYPES__ -ansi -Wshadow -Wno-write-strings -D_POSIX_SOURCE -Wall -Wextra -Wpedantic -Wcast-qual -Wcast-align -Wshadow -Wformat=2 -Wno-format-nonliteral -Wlogical-op -Wmissing-declarations -Wmissing-include-dirs -Wredundant-decls -Wswitch-default -Wundef -Wunused -Wunused-variable -Wunused-parameter -Wdisabled-optimization -Wnoexcept -Woverloaded-virtual -Wctor-dtor-privacy -Wnon-virtual-dtor -Wduplicated-cond -Wduplicated-branches -Wnull-dereference -Wuninitialized -Winit-self 
-- CMAKE_CXX_FLAGS:   -Wpointer-arith -Wcast-qual -D__USE_FIXED_PROTOTYPES__ -ansi -Wshadow -Wno-write-strings -D_POSIX_SOURCE -Wall -Wextra -Wpedantic -Wcast-qual -Wcast-align -Wshadow -Wformat=2 -Wno-format-nonliteral -Wlogical-op -Wmissing-declarations -Wmissing-include-dirs -Wredundant-decls -Wswitch-default -Wundef -Wunused -Wunused-variable -Wunused-parameter -Wdisabled-optimization -Wnoexcept -Woverloaded-virtual -Wctor-dtor-privacy -Wnon-virtual-dtor -Wduplicated-cond -Wduplicated-branches -Wnull-dereference -Wuninitialized -Winit-self  -DLINUX24_64 -DLINUX_X86_64
-- Checking VPR graphics option ON
-- VPR graphics is turned on, searching for dependencies
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tansell/github/LNIS-Projects/OpenFPGA/build
tansell@tansell-glaptop:~/github/LNIS-Projects/OpenFPGA/build$ cd ..
tansell@tansell-glaptop:~/github/LNIS-Projects/OpenFPGA$ mkdir build2
tansell@tansell-glaptop:~/github/LNIS-Projects/OpenFPGA$ cd build2/
tansell@tansell-glaptop:~/github/LNIS-Projects/OpenFPGA/build2$ cmake .. -DCMAKE_BUILD_TYPE=debug
-- CMAKE_BUILD_TYPE: debug
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpointer-arith
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpointer-arith - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-qual
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-qual - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-D__USE_FIXED_PROTOTYPES__
-- Performing Test CXX_COMPILER_SUPPORTS_-D__USE_FIXED_PROTOTYPES__ - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-ansi
-- Performing Test CXX_COMPILER_SUPPORTS_-ansi - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wshadow
-- Performing Test CXX_COMPILER_SUPPORTS_-Wshadow - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-allign
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-allign - Failed
-- Performing Test CXX_COMPILER_SUPPORTS_-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS_-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-D_POSIX_SOURCE
-- Performing Test CXX_COMPILER_SUPPORTS_-D_POSIX_SOURCE - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wall
-- Performing Test CXX_COMPILER_SUPPORTS_-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wextra
-- Performing Test CXX_COMPILER_SUPPORTS_-Wextra - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpedantic
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpedantic - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-align
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-align - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat=2 - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wno-format-nonliteral
-- Performing Test CXX_COMPILER_SUPPORTS_-Wno-format-nonliteral - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wlogical-op
-- Performing Test CXX_COMPILER_SUPPORTS_-Wlogical-op - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wmissing-declarations
-- Performing Test CXX_COMPILER_SUPPORTS_-Wmissing-declarations - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wmissing-include-dirs
-- Performing Test CXX_COMPILER_SUPPORTS_-Wmissing-include-dirs - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wredundant-decls
-- Performing Test CXX_COMPILER_SUPPORTS_-Wredundant-decls - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wswitch-default
[... 62 lines cut ...]
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found Readline: /usr/include  
-- Performing Test CXX_COMPILER_SUPPORTS_-w
-- Performing Test CXX_COMPILER_SUPPORTS_-w - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tansell/github/LNIS-Projects/OpenFPGA/build2
```

```
$ cat /etc/lsb-release 
DISTRIB_CODENAME=rodete
DISTRIB_DESCRIPTION="Debian GNU/Linux rodete"
DISTRIB_ID=Debian
DISTRIB_RELEASE=rodete
```

mithro, 2019-07-16:
```
$ gcc --version
gcc (Debian 7.3.0-18) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

mithro, 2019-07-16:
Note, I can successfully compile ACE2 in the Verilog-to-Routing repository....

tangxifan, 2019-07-16:
Hi, 
   We have just updated the master, which may solve your problem.
   I have tried using g++-7.3 in a CentOS system and it is working.
   Can you retry? We are glad to help if you have any issue.

mithro, 2019-07-16:
dev or master branch?

tangxifan, 2019-07-16:
master

mithro, 2019-07-16:
Still same issue;
```git
commit b0017cdbdf13be97b00b4749aee981d74fe3a930 (HEAD -> master, origin/master, origin/HEAD)
Merge: e2b7636 fe218fc
Author: AurelienAlacchi <42555907+AurelienUoU@users.noreply.github.com>
Date:   Tue Jul 16 13:08:24 2019 -0600

    Merge pull request #11 from LNIS-Projects/dev
    
    Dev
```

mithro, 2019-07-16:
Trying a clean build...

mithro, 2019-07-16:
clean build;
```
[100%] Linking CXX executable ../../abc/abc
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/bdd.c: In function ‘void ace_bdd_get_literals(Abc_Ntk_t*, st__table**, Vec_Ptr_t**)’:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/bdd.c:32:36: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
      (char*) Vec_PtrSize( *literals));
                                    ^
In file included from /home/tansell/github/LNIS-Projects/OpenFPGA/abc/src/misc/extra/extra.h:44:0,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/abc/src/base/abc/abc.h:42,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/ace.h:6,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/sim.c:2:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/sim.c: In function ‘void evaluate_circuit(Abc_Ntk_t*, Vec_Ptr_t*, int)’:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/sim.c:213:6: error: invalid static_cast from type ‘int*’ to type ‘int’
      assert(faninValues);
      ^
ace2/CMakeFiles/libace.dir/build.make:140: recipe for target 'ace2/CMakeFiles/libace.dir/SRC/sim.c.o' failed
make[2]: *** [ace2/CMakeFiles/libace.dir/SRC/sim.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/tansell/github/LNIS-Projects/OpenFPGA/abc/src/misc/extra/extra.h:44:0,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/abc/src/base/abc/abc.h:42,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/ace.h:6,
                 from /home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:3:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c: In function ‘void ace_io_print_activity(Abc_Ntk_t*, FILE*)’:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:37:3: error: invalid static_cast from type ‘Abc_Obj_t* {aka Abc_Obj_t_*}’ to type ‘int’
   assert(obj->pCopy);
   ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c: In function ‘int ace_io_read_activity(Abc_Ntk_t*, FILE*, ace_pi_format_t, double, double, const char*)’:
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:272:13: error: invalid static_cast from type ‘char*’ to type ‘int’
             assert(res);
             ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:295:17: error: invalid static_cast from type ‘char*’ to type ‘int’
                 assert(res);
                 ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:308:13: error: invalid static_cast from type ‘char*’ to type ‘int’
             assert(res);
             ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:311:17: error: invalid static_cast from type ‘char*’ to type ‘int’
                 assert(res);
                 ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:328:13: error: invalid static_cast from type ‘char*’ to type ‘int’
             assert(res);
             ^
/home/tansell/github/LNIS-Projects/OpenFPGA/ace2/SRC/io_ace.c:367:17: error: invalid static_cast from type ‘char*’ to type ‘int’
                 assert(res);
                 ^
ace2/CMakeFiles/libace.dir/build.make:127: recipe for target 'ace2/CMakeFiles/libace.dir/SRC/io_ace.c.o' failed
make[2]: *** [ace2/CMakeFiles/libace.dir/SRC/io_ace.c.o] Error 1
CMakeFiles/Makefile2:306: recipe for target 'ace2/CMakeFiles/libace.dir/all' failed
make[1]: *** [ace2/CMakeFiles/libace.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target abc
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

tangxifan, 2019-07-17:
Hi, 
   We have updated our ACE, being consistent with the VTR version.
   Can you retry?

   Best regards,
   Xifan Tang

closed 2019-12-02 by tangxifan