lnis-uofu/OpenFPGA #22: Functionnality request: Multiple verilog files given as benchmark

BaudouinChauviere, 2019-07-18:
It was requested by the University of Washington.

Pb: Most benchmarks or functional verilogs are actually projects with different files. Accepting multiple files would be a good step forward (also check if it is possible to use wild cards (*) or glob / similar function.

Comment: Need to check if it is possible to do under Yosys. This is going to be our limitations.

mithro, 2019-07-18:
@BaudouinChauviere We have used Multi-file flows with Yosys and VPR in SymbiFlow heavily. If you are at the ERI Summit I can discuss with you more about how to do this easily.

mithro, 2019-07-18:
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/yosys/synth.tcl

closed 2019-12-02 by tangxifan