lnis-uofu/OpenFPGA #42: Cmake build warning

aolofsson, 2020-01-22:
Probably non-critical...

Built from main. In Ubuntu 18.04 cmake gives the following error on missing X11. the regular dependancy installation attempts all failed to make a difference.

-- Could NOT find X11 (missing: X11) 
CMake Warning at vpr7_x2p/vpr/CMakeLists.txt:23 (message):
  Failed to find required X11 library (on debian/ubuntu try 'sudo apt-get
  install libx11-dev' to install)
-- Add flags to disable graphics in VPR compilation: -DNO_GRAPHICS

Not sure if this will be an issue downstream when running vpr?

A Makefile is produced and at first glance the build process seems ok...

Can you add a example/command sequence for how to test if install works in the README?

LNIS-Projects, 2020-01-23:
Hi,
   Thanks a lot for the feedback!
   The missing X11 will not cause compilation failure and neither any testing failure. 
   The GUI of VPR is mainly to help users to visualize how their designs are mapped on a FPGA.
   Without it, VPR will still implement your designs.

   Actually, the missing X11 is due to a version of Cmake which is not good at finding X11 libraries.
   We recommend users to try cmake 3.12 which we have validated.
   
    To address those issues, we have updated our README with more tips on compilation as well as guidelines about how to run quick test afterwards.
    
     Best regards,
     Xifan Tang

closed 2020-01-23 by LNIS-Projects

aolofsson, 2020-01-23:
Thanks Xifan! I used cmake 3.16.3. I am in the middle of testing 22 different projects and everyone is using different cmake versions:-( I figured it was safer to pick the latest stable one? Is the 3.12 actually better?

tangxifan, 2020-01-23:
Hi Andreas,
   In general the latest stable is the best choice. 
   Actually, we have tried cmake 3.13+ versions but they all have the issues on finding X11.

   However, as we are moving to vpr8 integration, this problem should be solved (VPR8 has changed to easyGL and GTK).
   I have tried vpr8 using cmake 3.16.3. GUI is working well.
   So, when our vpr8 integration is done. README should be updated and this restriction should be removed.

   Best regards,
   Xifan Tang

tangxifan, 2020-01-24:
Hi Andreas,
   We managed to remove this restriction on cmake today, thanks to @srtemp 
   We have updated our CMakeList script to have a better search for X11 library.
   Now, feel free to use the latest stable Cmake.
   It has been tested on our local machine as well as Travis.

   Best regards,
   Xifan Tang

aolofsson, 2020-01-24:
Hi Xifan, Thank you, that's awesome!