lnis-uofu/OpenFPGA #46: Hello, I have a OpenFPGA building question.

a3456410, 2020-02-19:
I try to build OpenFPGA, and cmake has a error result, this is content of the CMakeError.log file:

-----------------------------------------------------

Performing C++ SOURCE FILE Test CXX_COMPILER_SUPPORTS_-Wcast-allign failed with the following output:
Change Dir: /home/lab601/桌面/OpenFPGA-master/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_1ed51/fast 
/usr/bin/make -f CMakeFiles/cmTC_1ed51.dir/build.make CMakeFiles/cmTC_1ed51.dir/build
make[1]: Entering directory '/home/lab601/桌面/OpenFPGA-master/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_1ed51.dir/src.cxx.o
/usr/bin/c++    -DCXX_COMPILER_SUPPORTS_-Wcast-allign   -Wcast-allign -std=c++14 -o CMakeFiles/cmTC_1ed51.dir/src.cxx.o -c /home/lab601/桌面/OpenFPGA-master/build/CMakeFiles/CMakeTmp/src.cxx
c++: error: unrecognized command line option '-Wcast-allign'; did you mean '-Wcast-align'?
CMakeFiles/cmTC_1ed51.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_1ed51.dir/src.cxx.o' failed
make[1]: *** [CMakeFiles/cmTC_1ed51.dir/src.cxx.o] Error 1
make[1]: Leaving directory '/home/lab601/桌面/OpenFPGA-master/build/CMakeFiles/CMakeTmp'
Makefile:121: recipe for target 'cmTC_1ed51/fast' failed
make: *** [cmTC_1ed51/fast] Error 2

Source file was:
int main() { return 0; }

-----------------------------------------------------

is the command line  '-Wcast-allign' wrong? Or maybe the version of the compiler isn't right?

tangxifan, 2020-02-19:
Hello, 
     Can you provide more information about your system settings to help us understand the problem?
     We have provided detailed requirements on CMake and OS in [https://github.com/LNIS-Projects/OpenFPGA/blob/master/README.md](url).
      It looks like you are using Windows. Are you using GCC or VS?
    - If you are using GCC, make sure that your GCC version is 8+. Our continuous integration is using GCC 8.2.0. 
    - For VS, some flags may not be applicable. You find the setup in our CMAKE script [https://github.com/LNIS-Projects/OpenFPGA/blob/0cfb88a49f152aab0a06f309ff160f222bb51ed7/CMakeLists.txt#L76](url)
    I hope that can help you.
     Best regards,
     Xifan Tang

a3456410, 2020-02-20:
HI, I try to install some libraries in my UbuntuOS, and my building problem is solved! But thanks for your response.

closed 2020-02-20 by a3456410