lnis-uofu/OpenFPGA #47: I have some problem about running fpga flow.

a3456410, 2020-02-20:
I try to use python3 to run openfpga flow, but there are some error which I can't fix :
------------------------
lab601@lab601-VirtualBox:~/Desktop/OpenFPGA-master/openfpga_flow/scripts/test$ python3 ../run_fpga_flow.py k6_N10_sram_chain_HC_template.xml alu4.v
INFO - Validating commnad line arguments
INFO - Run directory : /home/lab601/Desktop/OpenFPGA-master/tmp
INFO - Running "yosys_vpr" Flow
INFO - Extracted lut_size size from arch XML = 6
INFO - Running Yosys with lut_size = 6
ERROR - Failed to run yosys
Traceback (most recent call last):
  File "../run_fpga_flow.py", line 454, in run_yosys_with_abc
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/lab601/Desktop/OpenFPGA-master/yosys/yosys', 'yosys.ys']' returned non-zero exit status 1.
ERROR - Current working directory : /home/lab601/Desktop/OpenFPGA-master/tmp
ERROR - 
ERROR - Exiting . . . . . .
---------------------------
I don't know what's wrong about yosys. If anyone can express their idea, I would appreciate it !

tangxifan, 2020-02-20:
Hi,
   Can you provide us the detailed log file generated by yosys?
   Instead of `run_fpga_flow.py`, I would suggest to use `run_fpga_task.py` which executes the `run_fpga_flow.py` in batch and has a nice user interface. 
   Both scripts have been documented in our documentation [https://openfpga.readthedocs.io/en/master/](url)
   And `run_fpga_task.py` is actually deployed in our continuous integration scripts [https://github.com/LNIS-Projects/OpenFPGA/blob/0cfb88a49f152aab0a06f309ff160f222bb51ed7/.travis/script.sh#L22](url).

   Best regards,
   Xifan Tang

a3456410, 2020-02-21:
Hi, thanks for your response!
There are two log files of yosys after executing `run_fpga_flow.py`, one is `yosys_output.txt`, another is `yosys.ys`. `yosys_output.txt` is a empty file, and the content of `yosys.ys` is following:

-----------

[yosys.ys.log](https://github.com/LNIS-Projects/OpenFPGA/files/4234949/yosys.ys.log)

-----------

And I also try to excute `run_fpga_task.py` using same benchmark (`alu4.v`) and same architecture (`k6_N10_sram_chain_HC_template.xml`). This time, yosys didn't have error messages, but vpr had a error message, which is `ERROR(1): Net #9 (o_0_) has no driver and will cause memory corruption.
`. I don't what's it meaning, and how to solve it.
This is content of the log file after executing `run_fpga_task.py` :

-----------

[00_alu4_MIN_ROUTE_CHAN_WIDTH_0_out.log](https://github.com/LNIS-Projects/OpenFPGA/files/4235025/00_alu4_MIN_ROUTE_CHAN_WIDTH_0_out.log)

-----------

And this is content of `vpr_stdout.log` : 

-----------

[vpr_stdout.log](https://github.com/LNIS-Projects/OpenFPGA/files/4235031/vpr_stdout.log)

-----------

tangxifan, 2020-02-24:
Hello,
  Thank you for the information.
  I looked into the report and I believe that the problem comes from that yosys version 0.9 produces some dangling nets that vpr7 cannot handle yet.
  As we are actively integrating to vpr8 and this problem should be solved very soon.
  For the current being, would you like to try yosys version 0.8 [https://github.com/YosysHQ/yosys/releases](url). 
  You just replace the yosys directory in OpenFPGA and that should solve your problem.
  Sorry for the inconvenience.
   
   Best regards,
   Xifan Tang

a3456410, 2020-02-25:
Thank you.
I tried to replace the directory of yosys version 0.8 in OpenFPGA and rebuilded the whole codes. OpenFPGA was builded successfully, but I used benchmark `alu4.v` to execute `run_fpga_task.py`, it still had the same error which said `ERROR(1): Net #9 (o_0_) has no driver and will cause memory corruption.`. I also tried another benchmark circuit `diffeq.v`, it still had a kindred error.
Therefore, I tried to use yosys version 0.7 and replace it in OpenFPGA, but the result of building OpenFPGA was failed. I guess the reason is because the version of adc is too new for yosys 0.7, maybe?
This is error message when I tried to build OpenFPGA which had yosys 0.7:

----------

[error_message.log](https://github.com/LNIS-Projects/OpenFPGA/files/4249418/error_message.log)

----------

tangxifan, 2020-03-01:
Hello,
   Can you tell me which alu4 you were trying? The one I am always using does not have a net `o_9_`.
[](https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/benchmarks/mcnc_big20/alu4/alu4.v)

    Best regards,
    Xifan Tang

a3456410, 2020-03-02:
Hi, The error which I had was at the net `o_0_`, not `o_9_`.
I totally tried three groups of the FPGA xml files and the benchmark verilog files, but they all had the same errors at the net `o_0_`. Those groups are:

-----

1.
FPGA arch: [https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/arch/template/k6_N10_sram_chain_HC_template.xml](url)
Benchmark: [https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/benchmarks/mcnc_big20/alu4/alu4.v](url)

-----

2.
FPGA arch: [https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/arch/template/k6_N10_sram_chain_HC_template.xml](url)
Benchmark: [https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/benchmarks/MCNC_Verilog/alu4/alu4.v](url)

-----

3.
FPGA arch: [https://github.com/LNIS-Projects/OpenFPGA/blob/master/fpga_flow/arch/template/k6_N10_sram_chain_HC_template.xml](url)
Benchmark: [https://github.com/LNIS-Projects/OpenFPGA/blob/master/fpga_flow/benchmarks/Verilog/MCNC/alu4/alu4.v](url)

-----

tangxifan, 2020-03-02:
Hello,
  Sorry. I messed up the net name. It is supposed to a logic synthesis problem. Trying different architecture files will not help.
  My understanding is that the Yosys script has not cleaned-up the netlist enough.

  Here are two possible solutions:
  - Debug with openfpga_flow scripts. When your flow errors out, all the intermediate files are kept in the ${OPENFPGA_ROOT}/openfpga_flow/task/<your_task_name>/latest/
     You can visit the directory and you will find an echo <your_task_name>_min_chan_width_vpr.txt, which contains the command how we call VPR.
     You can tweak the netlist and rerun the VPR.
  - Try This problem may be solved when using VPR8. Can you run the Yosys + VPR8 manually?
     If that solved the problem, I can help you using our latest openfpga with VPR8. 

     Best regards,
     Xifan Tang

a3456410, 2020-03-03:
Thanks for your advice!
I'm not sure how to modify the netlist which yosys generate, so I try to use it running VPR8. I executed VPR8 with `alu4.blif` yosys9 generated and `k6_N10_40nm.xml` VPR8 provided, and the result was successful. I also tried to modify a xml file OpenFPGA provided to use in VPR8, and thing wasn't going well. But I think the error problem may be solved if the version of VPR is the latest.

tangxifan, 2020-03-03:
Hello,
     Modifying the .blif netlist is not difficult. You can open the .blif netlist and search the net `o_0_`. If it is a dangling net, you should only see one matched result. Remove it and you should be good when running VPR.
    Our VPR8 version is still under test. If you still prefer trying VPR8, we can talk offline.

a3456410, 2020-03-04:
Hi,
I tried to modify `alu4.blif` which had the net error, but I found that the net `o_0_` didn't seem to be a dangling net. Following file is `alu4.blif` which isn't revised:

-----

[alu4.zip](https://github.com/LNIS-Projects/OpenFPGA/files/4285374/alu4.zip)

-----

In the .blif file which yosys generates, when a sentence change to the next line, the file will write a "\ " in the end of the sentence. I deleted all of the space characters which followed the "\" characters, reruned VPR, and the result was successful! So, if modify the code writing the .blif files which are converted from the .v files, the net error should be solved.

tangxifan, 2020-03-04:
Hello,
   Glad to know that the bug has been spotted. I believe this is a bug in Yosys BLIF writer. 
    More than you fixed in your repository, I would suggest you create a pull request on Yosys repository as well. [https://github.com/YosysHQ/yosys](url)

closed 2020-03-04 by tangxifan