lnis-uofu/OpenFPGA #48: Can OpenFPGA build a bidir segment FPGA architecture?

a3456410, 2020-03-16:
Hi, can OpenFPGA build a bidir segment FPGA architecture? 
I tried to change a undir segment FPGA, which was `k6_N10_sram_chain_HC_template.xml`, to a bidir segment by modifying the content of the `<segmentlist>` tag and the `<switchlist>` tag. This way was successful in VPR, but was failed in OpenFPGA. I tried to look for the example xml files of the bidir FPGA architectures, but I found no one. I also read the OpenFPGA documentation, but I couldn't find the content about describing a bidir FPGA architecture.

tangxifan, 2020-03-16:
Hello,
   Currently, we support uni-direction architecture mainly. If you try to run a bi-dir architecture, OpenFPGA will tell you clearly it is not yet supported. 
   This decision is based on the well-known paper [Directional and single-driver wires in FPGA interconnect](https://ieeexplore.ieee.org/document/1393249) 
   Majority of modern FPGAs use uni-directional routing architecture, which is far more efficient than the bi-directional.

   Best regards,
   Xifan Tang

a3456410, 2020-03-17:
Hi, Thanks for your response.
I am still interesting about describing a bidir structure in the OpenFPGA. If you can discuss it with me, I will be very grateful!

tangxifan, 2020-03-17:
Hello,
  Can you provide me an email address? So we can schedule a conference call? Thanks

closed 2020-03-17 by tangxifan