lnis-uofu/OpenFPGA #50: SDC global ports constraints

egiacomin, 2020-03-24:
Hello,

1- In global_ports.sdc, some global ports are defined as clocks while they clearly should not:

*****
create_clock Reset[0]-period 10 -waveform {0 5} [list [get_ports { Reset[0]}]]
set_drive 0 Reset[0]
******

Assuming the reset is assynchronous, I would suggest:

set_false_path -from [get_ports { Reset[0]}]

2- I would suggest defining the clock following the "standard" syntax:
create_clock -name prog_clk[0] -period 100 -waveform {0 50} [get_ports {prog_clk[0]}]
Instead of
create_clock prog_clk[0]-period 100 -waveform {0 50}

Thanks,
Edouard

tangxifan, 2020-03-25:
Should we still keep `set_drive 0`?

egiacomin, 2020-03-25:
I would also remove those as it will be technology dependent, and will also depend on how the FPGA externally, so it should be added manually by the physical design engineers.

closed 2020-03-25 by tangxifan