lnis-uofu/OpenFPGA #52: I try to run openfpga shell which using yosys_vpr flow of the task flow, and find some error.

a3456410, 2020-05-19:
Hi, I tried to run openfpga shell by using `run_fpga_task.py`. When the flow I chose was `vpr_blif`, the result was successful, but the flow I chose was` yosys_vpr`, the result was error. This is report about the reason of the error.

[00_FSM_top_MIN_ROUTE_CHAN_WIDTH_out.log](https://github.com/LNIS-Projects/OpenFPGA/files/4648500/00_FSM_top_MIN_ROUTE_CHAN_WIDTH_out.log)

My solution was using `yosys_vpr` flow to runing the section of yosys first, then generating `benchmark.blif`, `benchmark_ace_out.act`, `benchmark_output_verilog.v`. Finally, using those three files to run `vpr_blif` flow. This can be a temporary way to run a complete flow of openfpga shell.

ganeshgore, 2020-05-19:
`yosys_vpr ` is still not completely supported with `fpga_task `and `OpenFPGA shell`. In any case, can you share `task.conf` file

a3456410, 2020-05-19:
Sure, this is `task.conf` and my benchmark zip file.
[task.conf.log](https://github.com/LNIS-Projects/OpenFPGA/files/4649088/task.conf.log)
[FSM_three_code.zip](https://github.com/LNIS-Projects/OpenFPGA/files/4649093/FSM_three_code.zip)

closed 2020-07-22 by tangxifan