lnis-uofu/OpenFPGA #55: Segment fault

buddynohair, 2020-06-24:
Hello, i try to generate the Verilog netlists and testbenches with the command "python3 openfpga_flow/scripts/run_fpga_task.py openfpga_shell/configuration_chain" you have given on the toturial. 
But the application is crash because of a memory-access-error (core dumped) --> abgebrochen(Speicherabzug abgeschrieben)
[image]

tangxifan, 2020-06-24:
Hello,
    In my view, this is mainly due to the out-of-date path to the task file.
    Recently, the `configuration_chain` has been moved to `openfpga_shell/full_testbench/configuration_chain`.
    But the paths in the tutorial has not been updated.
    Meanwhile we are patching the tutorial, can you try 
    `python3 openfpga_flow/scripts/run_fpga_task.py openfpga_shell/full_testbench/configuration_chain`?

buddynohair, 2020-06-24:
Hello,
i don't think it's due to the path.  Actually i forgot to tell you that i already changed the commend with the correct path and receive this result, otherwise the file can't be found if i give the wrong path. 
[image]

tangxifan, 2020-06-24:
Hello,
    No worries. This test case is in our [CI](https://github.com/LNIS-Projects/OpenFPGA/blob/618c7d44c52114c4c247da01347007af084dd9ad/.travis/openfpga_vpr8_reg_test.sh#L15) which is tested on each commit.
   If it fails on a local machine, it is usually a matter of setup issue.    

    Would you mind sharing the `and2_run.openfpga` in your first screenshot?
    This will helps me reproduce the bug.

buddynohair, 2020-06-24:
hello,
i just ran the script with the new command 
cd ${TRAVIS_BUILD_DIR}
python3 /home/ukmsm/Schreibtisch/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py openfpga_shell/full_testbench/configuration_chain --debug --show_thread_logs
But i don't think the problem is solved.
[image]
[image]
[image]

tangxifan, 2020-06-24:
Hello,
   Can you provide more details so that we can reproduce your problem?
   For example: 
   - How did you compile OpenFPGA?
   - Which branch and which commit did you checkout?
   - What is the OS and what compilers you were using?

buddynohair, 2020-06-24:
Hello,
* I compiled the OpenFPGA with the following commends :
  git clone https://github.com/LNIS-Projects/OpenFPGA.git
  cd OpenFPGA
  mkdir build
  cd build
  cmake ..  -DCMAKE_BUILD_TYPE=debug
  make
* if i run the script manually, it says Assertion "!empty()" failed, core dumped as the first screenshot shows.
   if i run the script with the Commend ：
   python3 /home/ukmsm/Schreibtisch/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py    openfpga_shell/full_testbench/configuration_chain --debug --show_thread_logs. it says: 
  
 *subprocess.CalledProcessError: Command '['/home/ukmsm/Schreibtisch/OpenFPGA/openfpga/openfpga', '-f', 'and2_run.openfpga']' died with <Signals.SIGABRT: 6> 

  *File "/home/ukmsm/Schreibtisch/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 425, in run_single_script
  raise subprocess.CalledProcessError(0, " ".join(command))

  *subprocess.CalledProcessError: Command [...] returned non-zero exit status 0
   (As the third and fourth Screenshot shows)

*The system is Centos 8

tangxifan, 2020-06-24:
Hello,
    That may explain the problem. We are testing continuously on Ubuntu and Centos 7. 
    We have not tried Centos 8 yet.
    Can you share your `and2_run.openfpga` file? I would like see if there is anything that causes the seg fault.

buddynohair, 2020-06-24:
Hello,
yes for sure
[and2_run.openfpga.log](https://github.com/LNIS-Projects/OpenFPGA/files/4828192/and2_run.openfpga.log)

tangxifan, 2020-06-24:
Hello,
   The file seems to be ok. Can you run `valgrind openfpga -f and2_run.openfpga` to trace the segment fault?
   I will see where the bug comes from.

buddynohair, 2020-06-25:
Hello,
i have run it with the GDB and Valgrind. The first Screenshot is the result from GDB, and the rest is from Valgrind. im truly sorry, i don't know how to output the errors and save it as a log file.
My tutor has also run it with the GDB and has some ideas. For example: Error in libopenfpga/libopenfpgashell/src/shell.tpp, in line 300:
if ('#' == line.front()) {
#we need to test before, if the string is empty : if (line.empty() || '#' == line.front()) {

This could be a solution:

diff --git a/libopenfpga/libopenfpgashell/src/shell.tpp b/libopenfpga/libopenfpgashell/src/shell.tpp
index b5c506d4..47476ab6 100644
--- a/libopenfpga/libopenfpgashell/src/shell.tpp
+++ b/libopenfpga/libopenfpgashell/src/shell.tpp
@@ -297,7 +297,7 @@ void Shell<T>::run_script_mode(const char* script_file_name, T& context) {
   /* Read line by line */
   while (getline(fp, line)) {
     /* If the line that starts with '#', it is commented, we can skip */
-    if ('#' == line.front()) {
+    if (line.empty() || '#' == line.front()) {
       continue;
     }
     /* Try to split the line with '#', the string before '#' is the read command we want */
diff --git a/libopenfpga/libopenfpgautil/src/openfpga_digest.cpp b/libopenfpga/libopenfpgautil/src/openfpga_digest.cpp
index 00cbbe1e..f8eb3625 100644
--- a/libopenfpga/libopenfpgautil/src/openfpga_digest.cpp
+++ b/libopenfpga/libopenfpgautil/src/openfpga_digest.cpp
@@ -62,7 +62,7 @@ std::string format_dir_path(const std::string& dir_path_to_format) {
   std::replace(formatted_dir_path.begin(), formatted_dir_path.end(), illegal_back_slash, legal_back_slash);
 
   /* Add a back slash the string is not ended like this! */
-  if (legal_back_slash != formatted_dir_path.back()) {
+  if (formatted_dir_path.empty() || legal_back_slash != formatted_dir_path.back()) {
     formatted_dir_path.push_back(legal_back_slash);
   }
[image]

[image]
[image]
[image]

tangxifan, 2020-06-25:
Hello,
    Thanks for the inputs. Now we know the sources of the bug.
    I have patched on the master branch by following your suggestions.
    Please try it and see if it solves the problem.
  
    I also encourage you to create pull requests which can resolve the compatibility on different OS.

buddynohair, 2020-06-25:
Hello,
really thank you for your help. After this change, the application is not crash any more. I think we have dealt with the issue we talked before.
But there's new problem coming out. In the 00_and2/MIN_ROUTE_CHAN_WIDTH_out.log, it says: failed to execute iverilog_verification because of no such directory : iverilog.  And another error --> AttributeError: 'FileNotFoundError' object has no attribute 'output'
And i don't understand, in the openfpgashell.log, it says , the execution is successful finished.
[00_and2_MIN_ROUTE_CHAN_WIDTH_out.log](https://github.com/LNIS-Projects/OpenFPGA/files/4833701/00_and2_MIN_ROUTE_CHAN_WIDTH_out.log)
[openfpgashell.log](https://github.com/LNIS-Projects/OpenFPGA/files/4833706/openfpgashell.log)
[and2_run.openfpga.log](https://github.com/LNIS-Projects/OpenFPGA/files/4833718/and2_run.openfpga.log)

buddynohair, 2020-06-25:
Ah thank you , i just find the Verilog verification is the next step . Maybe i can run modelsim instead. And thank you again for solving the problem for me.

closed 2020-06-25 by LNIS-Projects

tangxifan, 2020-06-25:
> Hello,
> really thank you for your help. After this change, the application is not crash any more. I think we have dealt with the issue we talked before.
> But there's new problem coming out. In the [...]

Hello,
   Please make sure you have iVerilog correctly installed. You can find all the dependency required [here](https://openfpga.readthedocs.io/en/master/tutorials/compile.html#dependencies)