lnis-uofu/OpenFPGA #76: user_defined_templates.v

buddynohair, 2020-08-06:
Hi,
i got a new problem.  
After i use commend "python3 openfpga_flow/scripts/run_fpga_task.py openfpga_shell/full_testbench/configuration_chain" to generate the verilog code, i find the module in the file "user_defined_templates.v" is not defined (Screenshot 1), although it was already described in the architecture xml.file. (Screenshot 2).
[image]
[image]

tangxifan, 2020-08-06:
Hello,
   Thank you for trying OpenFPGA. The `user_defined_templated.v` is intentionally created with no logic functionality inside.
   It aims to help engineers to check what is the port sequence required by top-level Verilog netlists (when you use implicit port mapping).
   As you define these circuit models to be external, we expect you use the `user_defined_templated.v`  as a reference to check the port sequence in your Verilog modules.
 
   You can find detailed documentation here:

   [online documentation about write_fabric_verilog command](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands.html#write-fabric-verilog)
   
   [online documentation about the Verilog file](https://openfpga.readthedocs.io/en/master/manual/fpga_verilog/fabric_netlist.html#cmdoption-arg-user_defined_templates.v)

buddynohair, 2020-08-06:
Hello,
really thank you for your prompt reply. So i get one more question. If i want to generate a customized module（instance） under the OpenFPGA framework , is it possible to achieve that or i have to write it manually in the user_defined_templated.v .

tangxifan, 2020-08-06:
Hello,
   If you have a customized module to be used by OpenFPGA, as you defined in your architecture file shown in the screenshot 2, you have to provide your Verilog modules. Since you specify an external netlist in the circuit model definition, OpenFPGA will not auto-generate these files.
  There are several ways:
  - Write your own Verilog modules, such as inverter, buffer etc., to a Verilog netlist. Then, you specify the file in the circuit model definition by using `verilog_netlist` or `spice_netlist`.
  - Use the user-defined template files and fill the content yourself. Afterward, you specify the file in the circuit model definition by using `verilog_netlist` or `spice_netlist`.
  - you specify the standard cell library (a Verilog netlist) in the circuit model definition by using `verilog_netlist` or `spice_netlist`.

  You can see our [example](https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/mux_design/stdcell_mux2/config/task.conf)

buddynohair, 2020-08-06:
Hello, 
Im sorry. i didn't understand this step "specify the file in the circuit model definition by using verilog_netlist or spice_netlist." correctly . What should i actually do with this step?
[image]
[image]

i used your 1bit Adder for example. you can see through the screenshot, i have already filled it in the architecture xml.file, but there's still no logic function.

tangxifan, 2020-08-06:
Hello,
   It is basically similar to what you show in the screenshot.
   You need to specify the full path to your Verilog netlists because there are external netlists (OpenFPGA will not generate them). This is an [example](https://github.com/LNIS-Projects/OpenFPGA/blob/39c6441b73cd6f802d81a04f977d619bae415058/openfpga_flow/openfpga_arch/k6_frac_N10_stdcell_mux_40nm_openfpga.xml#L89).
   
   As this is defined as an external netlist, OpenFPGA will not generate any content as our documentation clarifies. It is expected that the `user_defined_template.v` does not include logic function, other than an entity.
   It means that you must modify this netlist upon your own needs before using it in OpenFPGA.

buddynohair, 2020-08-06:
Hello,
i got it. Thank you for your help.

closed 2020-08-06 by buddynohair