lnis-uofu/OpenFPGA #80: pb_type missing circuit binding

buddynohair, 2020-09-02:
Hi,

this time i wrote a new custom block called "blackbox", which includes an adder and a flipflop module in the vpr-arch, and want to link it with the circuit model in the openfppa-arch. Unfortunately it shows me an error called "pb_type missing circuit binding". I checked pb_type annotation part with the tutorial and the examples you have given, but didn't find any problems. 
Can you please give me some advice?

[image]

[xml.zip](https://github.com/LNIS-Projects/OpenFPGA/files/5164827/xml.zip)

tangxifan, 2020-09-02:
Hello,
   I checked your architecture file. It is due to your binding lines
   `<pb_type name="blackbox.[adder].ble.adder">`
   The name follows the naming in vpr on the pb_type name in full hierarchy.
   The format is as follows.
    `<pb_type_name>[<mode_name>].<sub_pb_type_name>[<mode_name>]`
   Therefore, the dot between `blackbox` and `[adder]`  should not be there.
   I believe that is the issue.

buddynohair, 2020-09-02:
Hi， 
i delete the dot between blackbox and [adder] , but the issue still exists 
[image]

tangxifan, 2020-09-03:
Hello,
    After checking your openfpga architecture XML, I see another problem.
    You define the pb_type-to-circuit_model binding inside the  `<pb_type name="blackbox">`, which will be discarded by the parser. 
    As you can see in the examples, the binding should be defined under the `<pb_type_annotation>` without any intermediate hierarchy.
    Also, I think the debugging information may be too brief to help. Therefore I created a pull request where the full hierarchy path of a pb_type will be printed out when the binding errors out.

buddynohair, 2020-09-03:
Hello, 
im still confused with the pb_type_annotation format.
I simplify the block blackbox, now only with one adder module. I also change the binding line in the openfpga-arc as followings 
[image]

But it still shows me the error pb_type "adder" missing circuit model binding.

[xml.zip](https://github.com/LNIS-Projects/OpenFPGA/files/5169634/xml.zip)

tangxifan, 2020-09-03:
Hello,
   What you show here is not consistent with your XML file:
   `<pb_type name="blackbox[adder].ble.adder" circuit_model_name="adder"/>`
   When I change to this line to what you showed, OpenFPGA works.

buddynohair, 2020-09-03:
Hello,
thank you very much. I forgot to save the file before i ran the commend.

closed 2020-09-03 by buddynohair