lnis-uofu/OpenFPGA #82: bitstream generation for custom block

buddynohair, 2020-09-03:
Hello,
im now working on the bitstream generation for the new custom block, for example blackbox. 
According to your paper TGA+19, i generate a physical mode and an operating mode, both have the module adder(because there's only one module: adder in the blackbox). Afterwards i combine them in the openfpga-arc.
But the issue is that the bitstream isn't changed. It still only has "clb" block and "swich" block, what means the bitstream for blackbox isn't generated.
can you please give me some advice about that?
[k4_N4_.zip](https://github.com/LNIS-Projects/OpenFPGA/files/5171713/k4_N4_.zip)

tangxifan, 2020-09-03:
Hello,
   In your VPR architecture file, you defined the pb_type `blackbox` at the same level as `clb`, it means the `blackbox` will be an  physical tile but not an operating mode of `clb`. In your `blackbox`, it is 100% hard logic. There is no LUT and programmable routing. As a result, you see no changes in the bitstream.
   I would suggest you to look into our multi-mode examples at
   [k6_frac_N10_tileable_adder_chain_40nm](https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/vpr_arch/k6_frac_N10_tileable_adder_chain_40nm.xml)
   In this directory, you can find around 10 examples of multi-mode CLBs, BRAMs and DSPs.
   Also in our regression tests, you can easy run these [examples](https://github.com/LNIS-Projects/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/hard_adder/config/task.conf)

buddynohair, 2020-09-04:
您好，
实在不好意思，大佬，可不可以用中文说一下。
是这样的，我们学校自己有块板子，并且想把板子移植到你的框架上来。但是我们学校板子上有很多新的逻辑器件逻辑功能，所以想先通过您的框架生成一个一个带有新的逻辑功能的新的模块，以此先尝试一下，查看这是否是可行的。我看了您的例子
，k6_frac_N10_tileable_adder_chain_mem16K_40nm.xml 这边您创建了一个新的bram模块但
是我发现在bitstream那一块并没有为bram那一块生成，而且在routing那一块我也只看到了clb那一部分，并没有找到bram那一部分。所以我想请问一下，bitstream可不可以为除了clb以外的其他板块生成，并且有没有可能也为新的模块布线？

tangxifan, 2020-09-04:
你好， 这个问题我们可以私下邮件交流。我的邮箱是xifan.tang@utah.edu

closed 2020-09-04 by tangxifan