lnis-uofu/OpenFPGA #99: CMake missing libffi-dev in Ubuntu 20.04

lukefahr, 2020-10-02:
## Problem 

```
frontends/ast/dpicall.cc:25:10: fatal error: ffi.h: No such file or directory
   25 | #include <ffi.h>
      |          ^~~~~~~
compilation terminated.
```

## Steps
```
git clone https://github.com/LNIS-Projects/OpenFPGA.git
cd OpenFPGA
mkdir build; cd build
cmake .. 
make  
```

## Solution

I think your Cmake is missing a required package, at least for newer version of Ubuntu.  I solved the issue with:

`sudo apt-get install libffi-dev`

## Envionment 

`$ lsb_release -a `

```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

tangxifan, 2020-10-02:
Hi, Thanks for trying OpenFPGA. I believe this is a valid bug. The 'lib-ffi' package is required by yosys but the current CMake scripts for yosys is simply a wrapper for yosys makefile. I will improve the CMakefile in a pull request.

tangxifan, 2022-09-06:
Addressed in the current master already.

https://github.com/lnis-uofu/OpenFPGA/blob/85532dd12918d97179ddc732c8e15902523a5bcd/.github/workflows/install_dependencies_build.sh#L19

Close this issue.