lnis-uofu/OpenFPGA #120: A small request to the tool creator

GunnerG, 2020-11-08:
hey guys,
hiii
can you guys make a **screencast/video series** for the detailed usage of this tool..?
i know there is detailed documentation but still if we get a video series that would be really great..
sorry for my poor english..
cheers...

tangxifan, 2020-11-12:
Hello,
   Thank you for your interests in OpenFPGA. I believe that video tutorials can bring a great value to OpenFPGA and beginner-level users.  We are planning to create a few in near future. On the other sides, we have a few tutorials in the documentation, for instance:
-  [A quick start on OpenFPGA design flow](https://openfpga.readthedocs.io/en/master/tutorials/design_flow/index.html)
-  [A quick start on Customizing Architecture](https://openfpga.readthedocs.io/en/master/tutorials/arch_modeling/index.html).
   After trying these tutorials, I believe you have learned the fundamental knowledges about how to use the tools.

   If you have any questions, feel free to ask.

tangxifan, 2022-09-06:
Close the issue as answered. If you need further help, please comment (reopen this issue) or create a new issue.
Thank you for the interests in OpenFPGA.

closed 2022-09-06 by tangxifan