lnis-uofu/OpenFPGA #125: Failed to extract lut_size from XML file

GunnerG, 2020-11-13:
**To Reproduce**
Steps to reproduce the behavior:
1]Make a directory  in scripts folder called "test"
2] ~/Desktop/OpenFPGA/openfpga_flow/scripts/test$ python3 ../run_fpga_flow.py 
    ../../openfpga_arch/k4_frac_N4_adder_chain_mem1K_40nm_frame_openfpga.xml 
    ../../benchmarks/micro_benchmark/and2/and2.blif --top_module and2.v

**Screenshots**
[image]

**Enviornment (please complete the following information):**
anesh@ganesh:istributor ID:	Linuxmint
Description:	Linux Mint 20
Release:	          20
Codename:	ulyana
gcc_version:    gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

**Additional context**
Add any other context about the problem here.

tangxifan, 2020-11-13:
Hello,
   Can you let me know which tutorial you are following?

GunnerG, 2020-11-14:
Hello,
i am following these tutorials ----> 

- https://openfpga.readthedocs.io/en/master/tutorials/design_flow/index.html

- https://openfpga.readthedocs.io/en/master/tutorials/arch_modeling/index.html

tangxifan, 2020-11-14:
Hello,
   If you were running the example from tutorials. You should use the openfpga_task.py rather than openfpga_flow.py
   For example:
   `python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/full_testbench/configuration_chain`

   Also, if you see errors, you can turn on the debug mode: 

   `python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/full_testbench/configuration_chain --debug --show_thread_logs`

   If you are confused by error messages, feel free to ask.

tangxifan, 2022-09-06:
Close the issue as answered. If you need further help, please comment (reopen this issue) or create a new issue.
Thank you for the interests in OpenFPGA.

closed 2022-09-06 by tangxifan