lnis-uofu/OpenFPGA #145: Issue adding Yosys support for new device

lnsharma, 2020-12-03:
I noticed multiple issues in the current task run flow. It looks to have various hard-codings or assumptions.
My intention is to write a testcase to run yosys by invoking synth_quicklogic and here are the issues/challenges I faced:

1. File to run a task - task.conf expects only these 4 synthesis parameters:
- Top_module
- Chan_width
- Act
- Verilog

2. The issue with above approach is that it expects yosys to run only with these variables. I gone through run_fpga_flow.py script:
- Looks like this script has a hardcoded yosys template file name - ys_tmpl_yosys_vpr_flow.ys
- It just picks the above 4 variables and put it in this file. Then this file is used for yosys run.

3. I think the current approach may not run for yosys cmdline options like: synth_quicklogic. Similar cmdline options are there for other devices also in yosys. I think we need to update this script to support this kind of variables.

4. Moreover, with task.conf approach, I was not even able to run yosys. So, I ran the script in the following manner:

- python3 ${OPENFPGA_PATH}/openfpga_flow/scripts/run_fpga_flow.py ${OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_frac_N4_tileable_40nm.xml ${OPENFPGA_PATH}//openfpga_flow/openfpga_arch/k4_frac_N4_40nm_cc_openfpga.xml --top_module counter --run_dir tmp --yosys_tmp counter.ys

5. Please note that in the above commandline, even if I provided option “—yosys_tmp” but the script does not use this option at all to take the template file as input, rather it always picks “ys_tmpl_yosys_vpr_flow.ys”

ganeshgore, 2020-12-03:
Oh that is true, let me fix quick.

ganeshgore, 2020-12-04:
Yes the latest commit fixes the yosys template bug

lnsharma, 2021-01-11:
Issue is fixed now.

closed 2021-01-11 by lnsharma

abdullahyildiz, 2021-03-18:
Does OpenFPGA support generating techmap file from XML to be used in Yosys?

tangxifan, 2021-03-19:
@abdullahyildiz Currently we do not have such support in OpenFPGA. It is on our to-do list.