lnis-uofu/OpenFPGA #153: Adopt yosys-symbiflow-plugins for SDC support?

mithro, 2020-12-08:
You might want to consider adopting the usage of the SymbiFlow Yosys Plugins @ https://github.com/SymbiFlow/yosys-symbiflow-plugins to improve your SDC support (and future timing driven synthesis). This works by importing the SDC into Yosys cells and then writing out a modified SDC at the end. This means if Yosys changes module names or optimizes out your modules you still end up with a valid SDC file you can pass to VPR.

tangxifan, 2020-12-08:
@mithro I think it is a good idea. Is this effort will be merged to yosys master ? We are now adding yosys as a submodule (see pull request #152 ) and it will be very easy to integrate the latest Yosys. 

Otherwise,  I will find a way to integrate this feature.

mithro, 2020-12-08:
It's an external plugin for Yosys, so it should work with any new Yosys version.

tangxifan, 2020-12-10:
Bring here the note from #152 :
A follow-up pull request is needed which consider the benchmark [https://github.com/smunaut/iua](https://github.com/smunaut/iua) as a test. That needs SDC to specify the minimum clock requirements to meet USB 2.0 speed.

tangxifan, 2021-12-17:
https://github.com/SymbiFlow/yosys-symbiflow-plugins is now a submodule of OpenFPGA.
Dependabot has been set up to auto tracking latest updates.

So I close this issue. If you have any question, feel free to propose.

closed 2021-12-17 by tangxifan