lnis-uofu/OpenFPGA #171: OpenFPGA flow parsing LUT INIT params

lnsharma, 2021-01-11:
**Is your feature request related to a problem? Please describe.**
During adder implementation, if yosys has to synthesise an adder as a hard-macro and then in OpenFPGA flow, this is implemented as LUT+carry_follower where LUT can be a combination of LUT4 with LUT2s or LUT3 with LUT2s. As yosys is implementing it as a hard-macro and it comes as a subckt in openfpga with INIT params that is an alternate to truth table that generally present in blif when luts are represented as .names. 

**Describe the solution you'd like**
OpenFPGA should enable parsing parameters that has information similar to what a truth table provides to .names instances in blif

**Describe alternatives you've considered**
Alternate solution considered is to implement adder module in yosys in terms of LUT3+LUT2s+carry_follower. However, there is no guarantee in vpr that packers packs them all together so it may not provide a robust solution.

**Additional context**
Add any other context or screenshots about the feature request here.

tangxifan, 2021-01-11:
Just checked VPR's [blif parser](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/base/read_blif.cpp). It seems that `.init` is not supported yet. 
Then, it means that we may have to consider a different file format to take the `.init` parameters.

lnsharma, 2021-01-12:
I think we can take either of these two approaches:

1. Refer to [https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#vpr-eblif-cname](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#vpr-eblif-cname): .eblif (Extended BLIF) format supports both .param and .attr on subckt. This can enable us to specify the parameters.
2. Refer to section 6 of doc: [https://docs.verilogtorouting.org/en/latest/_downloads/a79b6634b582f56c053f2abad923112a/blif.pdf](https://docs.verilogtorouting.org/en/latest/_downloads/a79b6634b582f56c053f2abad923112a/blif.pdf): Need to write model-reference in the generated blif file. Currently, blif just add subcircuit instantiations in blif but no definition of it. For example, generated blif has these lines:

_.model pipelined_8bit_adder
.inputs clk ren wen raddr_0_ raddr_1_ raddr_2_ raddr_3_ raddr_4_ raddr_5_ waddr_0_ waddr_1_ waddr_2_ waddr_3_ waddr_4_ waddr_5_ a_0_ a_1_ a_2_ a_3_ a_4_ a_5_ a_6_ b_0_ b_1_ b_2_ b_3_ b_4_ b_5_ b_6_
.outputs q_0_ q_1_ q_2_ q_3_ q_4_ q_5_ q_6_ q_7_
.names $false
.names $true
1
.names $undef
.subckt soft_adder A=a_st1[6] B=b_st1[6] CI=AplusB_soft_adder_O_CO[5] CO=AplusB_soft_adder_O_CO[6] I3=$false O=AplusB[6]
.subckt soft_adder A=a_st1[5] B=b_st1[5] CI=AplusB_soft_adder_O_CO[4] CO=AplusB_soft_adder_O_CO[5] I3=$false O=AplusB[5]_

However, there is no definition of subckt - soft_adder, if we can dump even soft_adder definition in terms of .names and another subckt for carry in the same blif, then openfpga can get the truth table of the luts used in soft_adder.

closed 2021-06-29 by tangxifan