lnis-uofu/OpenFPGA #181: Move OpenFPGA docs to github pages

ganeshgore, 2021-01-23:
**Is your feature request related to a problem? Please describe.**
Move doc building to gh-pages than readthedocs, because 
1. reduces the dependencies on other services 
2. gh-pages are more customizable

**Describe the solution you'd like**
Create a docker image to build docs on the master branch and deploy on dedicated gh-pages branch

**Describe alternatives you've considered**
Seems feasible to achieve above thing 

**Additional context**
It will be great if we could build docs for push requests as well (if there are any changes) and have it hosted in the same github pages. It will make documentation review much faster.

tangxifan, 2021-01-23:
Currently, readthedocs is still working well. We can preview the documentation for any pull request: 

[image]

So, I will hold this issue until that readthedocs can no longer fit our needs.

mithro, 2021-01-24:
I would highly recommend sticking with readthedocs.

closed 2021-04-24 by tangxifan