lnis-uofu/OpenFPGA #183: Introduce -b batch mode execution flag in openfpga shell

ganeshgore, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
Currently in case of fatal error during openfpga_shell script execution the shell aborts execution and enters in interactive mode. During the regression test (which runs in a non interactive mode) the fatal error freezes the script.

**Describe the solution you'd like**
We need to have `-b` batch mode command like flag. 
Which will terminate the openfpga shell with correct exit status code instead of entering in interactive mode.

**Describe alternatives you've considered**
The above solution is feasible

**Additional context**
You can regenerate this behavior locally with wrong command argument.
Example `build_fabric --compress_routing1`

ganeshgore, 2021-01-26:
`-batch` worked well. This can be closed if we add `-v` flag as well.

tangxifan, 2021-01-26:
-v flag is for version display, right? If so, we should reopen that issue and close this one.

closed 2021-01-27 by tangxifan