lnis-uofu/OpenFPGA #198: Regression tests not run for yosys submodule updates

tangxifan, 2021-01-27:
**Describe the bug**
Regression tests in CI are skipped when there is an update in yosys submodule 

**To Reproduce**
See the screenshot attached. 

**Expected behavior**
Regression tests should be invoked in CI when there is a change in the yosys submodule.
Otherwise, there is no guarantee that OpenFPGA can be functional.

**Screenshots**
[image]

**Enviornment (please complete the following information):**
 - OS: CentOs
 - Compiler: gcc
 - Version: latest master

tangxifan, 2021-01-27:
@ganeshgore I would like to know why it happened and how soon we can address it.

closed 2021-01-27 by tangxifan

ganeshgore, 2021-01-27:
I will check this and fix, today