lnis-uofu/OpenFPGA #209: OpenFPGA installation to shared non-user writable loxaiton

nachiket, 2021-02-01:
From what I understand the OpenFPGA run_task_flow.py scripts write their run*** outputs in the task configuration folders located inside the OpenFPGA directories. 

I need to be able to supply my own configuration from a separate location and have the tool flow use a folder outside OpenFPGA install directory for its outputs.

This is because the tool will be installed in /opt or some globally visible path in a multi-user environment. I’d rather not force everyone to clone the full repository.

Am I missing a way to do this?

tangxifan, 2021-02-01:
Thank you for your interests in OpenFPGA. @ganeshgore I believe this feature is supported by openfpga flow scripts. Can you provide more details for @nachiket ? 
We can build a tutorial from this effort and add it to our documentation. @nachiket is that helpful?

nachiket, 2021-02-01:
Sounds good -- it would help to have a multi-user installation workflow that doesn't need everyone to have write access into the OpenFPGA installation folder.

nachiket, 2021-02-01:
Here's the command I'm running:
```
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py basic_tests/generate_fabric
```

I'd like the run** folders to be generate outside OPENFPGA_ROOT. I will provide and edited `generate_fabric` configuration myself without using the version in OPENFPGA_ROOT.

Here OPENFPGA_ROOT=/opt/OpenFPGA where I've installed OpenFPGA.

ganeshgore, 2021-02-01:
Hello Nachiket,

Thank you for using OpenFPGA. We also have a similar multi-user environment, and this is how we generally integrate with OpenFPGA.

Create a directory `MY_OPENFPGA_TASK` consisting of `config/task.conf` and `runOpenFPGA.sh` anywhere in your environment.

The directory structure will look like this 
```
<directory XYZ>
├── runOpenFPGA.sh
└── MY_OPENFPGA_TASK
        ├── config
             └── task.conf
```

```bash
# runOpenFPGA.sh
export OPENFPGA_PATH=<Root directory of OpenFPGA path>
export TASK_NAME=MY_OPENFPGA_TASK

rm -f ${OPENFPGA_PATH}/openfpga_flow/tasks/${TASK_NAME}

ln -s ${PWD}/${TASK_NAME} ${OPENFPGA_PATH}/openfpga_flow/tasks/

(cd $OPENFPGA_PATH && source openfpga.sh && run-task ${TASK_DIR_NAME})
```
Run `runOpenFPGA.sh` script from  the same directory XYZ.
In case you have any supporting files like `.xml`, `.openfpga` or `.v` you can add them under the MY_OPENFPGA_TASK directory  and use relative paths.

let me know if that helps.

nachiket, 2021-02-01:
I don't have write permissions into the OPENFPGA folder. The `rm` and `ln` commands above won't work.

ganeshgore, 2021-02-01:
Ok let me check few things and get back to you asap.

nachiket, 2021-02-08:
Hi -- do you have a suggested workflow for this operation mode?

ganeshgore, 2021-02-08:
Hello, I created a pull request yesterday #221, It will be merged today after the review.
Now task script will support this operating mode.

```
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py generate_fabric
```
where `generate_fabric` is a task in the current directory.

Please let me know if that addresses your problem

nachiket, 2021-02-08:
Ok I'll wait for the merge and test it out today. Thanks a lot!

nachiket, 2021-02-08:
I just tried it manually and seems to work fine. Thanks!

closed 2021-02-08 by nachiket

nachiket, 2023-09-25:
Commented on another thread, but just checking-in on this thread to see if this is supported in the newer OpenFPGA release. It seem to break the 2021 teachings scripts.

tangxifan, 2023-09-25:
@nachiket We have reworked the flow scripts to ease the use w.r.t. the non-user writable location. Now you do not need to specify any local task. You can directly call the script by providing the absolute path to your task folder.

For example,
 
```
python3 openfpga_flow/scripts/run_fpga_task.py /home/xifan/example_task
```

nachiket, 2023-09-25:
Perfect! That worked.

Just and FYI, relative path didn’t seem to work.