lnis-uofu/OpenFPGA #225: Standard-cell flow

nachiket, 2021-02-08:
I'm trying to work out how to setup a standard cell flow and provide my own implementations for the SRAM call and Pass transistor. For a simple `generate_fabric` run, I can do this via the following patch file `patch.txt` to the Verilog. Can someone please check if this is adequate, or tell me how I can confirm correctness of the patched design? The Synopsys DC tool seems to compile the design just fine with these fixes.

[patch.txt](https://github.com/lnis-uofu/OpenFPGA/files/5944885/patch.txt)

tangxifan, 2021-02-08:
@nachiket I have seen similar problems from other users who are exploiting a pure standard cell flow. 
The suggested solution is not to patch the auto-generated netlists but let openfpga to use external Verilog netlists (see the XML syntax ``verilog_netlist`` [here](https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_library/#cmdoption-arg-spice_netlist)).

So, you can create your own Verilog modeling for inverters etc. and OpenFPGA can include them in the auto-generated netlists.

About how to verify the netlists, you can take a look at the [tutorial ](https://openfpga.readthedocs.io/en/master/tutorials/design_flow/verilog2verification/)
The ``generate_fabric`` test cases focus on netlist generation only. You can look into other test case (e.g. the one in the tutorial) where you can run a verification by the end of the flow.

nachiket, 2021-02-08:
I'm a little overwhelmed :P Do you have a list of components that would need to be adapted for a standard cell flow? I just did a trial-and-error patching on the generated Verilog until it passed synthesis. For external Verilog netlists, is there a standard cell compatible version already available I could reuse?

tangxifan, 2021-02-08:
@nachiket No worries. These are valid bugs and actions should be taken to make these primitive modules synthesizable. 
As for your questions,

> Do you have a list of components that would need to be adapted for a standard cell flow? 
- The circuit models relevant to these problems are the ``inverter``, ``buffer``, ``pass_gate_logic``. 

> For external Verilog netlists, is there a standard cell compatible version already available I could reuse?
- I just checked the openfpga cell library. We do not have these netlists. But I believe that we should create these netlists. if you have time, I can help you in this effort. And we can create a test case and a tutorial from this effort.

Please let me know what you think.

tangxifan, 2021-02-08:
@nachiket No worries. Let me give a detailed step-by-step tutorial here.
- Copy the test case [std_mux ](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/mux_design/stdcell_mux2/config/task.conf), which is close to the synthesizable FPGA fabric (built with standard cells while majority test cases are built with custom cells). You can create a testcase ``synthesizable_verilog`` under [fpga_verilog](https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/tasks/fpga_verilog) directory.
- Verification is applicable to synthesizable Verilog only when using the full testbenches. So you need to comment out [LINE 37](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/tasks/fpga_verilog/mux_design/stdcell_mux2/config/task.conf#L37) at the task configuration file
- You can add synthesizable Verilog netlists, such as ``inv.v`` and ``buf.v`` to the [openfpga_cell_library directory](https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/openfpga_cell_library). The content of these netlists can be directly copied from your patched netlists. Also remember to create a cell for the DFFSRQ based on your patch. You can probably rename to another cell name, e.g., ``DFFSRQ_SYTHN``
- You then need to duplicate the [openfpga architecture file](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/openfpga_arch/k6_frac_N10_stdcell_mux_40nm_openfpga.xml) used by this test case. Explicitly, you need to specify ``verilog_netlist`` for the circuit models ``INVTX1``, ``buf4`` etc,  using the Verilog netlists created in step 2. You can refer to [this example](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/openfpga_arch/k6_frac_N10_stdcell_mux_40nm_openfpga.xml#L89).
- After these step, we can run openfpga_flow for the new test case.

If you have any problems about the steps, feel free to ask.

nachiket, 2021-02-09:
So I did most of these steps, and I get the following error:

```
ERROR - iverilog_verification run failed with returncode 1
ERROR - command iverilog -o compiled_and2 ./SRC/and2_include_netlists.v -s and2_autocheck_top_tb
ERROR - -->>error: Unable to find the root module "and2_autocheck_top_tb" in the Verilog source.
ERROR - -->>1 error(s) during elaboration.
```

Once I get this working, how do I submit a patch with these fixes? I’ve pushed my changes to my fork here: https://github.com/watcag/OpenFPGA <https://github.com/watcag/OpenFPGA>

Nachiket

tangxifan, 2021-02-09:
@nachiket I checked out your branch and run your task quickly. It seems that the error parsing does not give you the full details. Here is what I did when the flow-run fails:

- Go the runtime directory, which is in the ``latest`` directory under your test case. In you case, it is ``openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/latest/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH``
- Since you failed in running iverilog, you can quickly checkout the file ``iverilog_output.txt`` to find out the full messages. Here are what I see: 

```
././SRC/fabric_netlists.v:17: Include file /var/tmp/xtang/OpenFPGA/openfpga_flow/openfpga_cell_library/verilog/buf.v not found
error: Unable to find the root module "and2_autocheck_top_tb" in the Verilog source.
     : Perhaps ``-s and2_autocheck_top_tb'' is incorrect?
1 error(s) during elaboration.
```

This is due to in your openfpga arch XML, you set the wrong file path for circuit model buf4, which is supposed to be ``buf4.v``.

You can keep debugging by running the iverilog until it compiles well. According to what I see, iVerilog complains about 

```
/var/tmp/xtang/OpenFPGA/openfpga_flow/openfpga_cell_library/verilog/dffsrq.v:28: syntax error
/var/tmp/xtang/OpenFPGA/openfpga_flow/openfpga_cell_library/verilog/dffsrq.v:29: Syntax in assignment statement l-value.
```

Once the test case passed verification, can you create a pull request? I will be happy to review it and merge to master.

nachiket, 2021-02-09:
Never mind — I had a typo in the XML config file and a bug in my Verilog for DFF. However, things seem to be stuck endlessly somewhere here:

```
$ python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py synthesizable_verilog
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task synthesizable_verilog
 INFO (     MainThread) - Created "run003" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs

```

nachiket, 2021-02-09:
`top` tells me that `vvp` process is running at 100% CPU for the last few minutes.

nachiket, 2021-02-09:
How long does `vvp verification` typically take?

I killed the program and ran the full python command as indicated by the trace
```
python3 /opt/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /opt/OpenFPGA/openfpga_flow/vpr_arch/k6_frac_N10_tileable_40nm.xml /opt/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /spare/opt/OpenFPGA/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/run001/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /opt/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /opt/OpenFPGA/openfpga_flow/openfpga_arch/k6_frac_N10_stdcell_mux_40nm_openfpga_synthesizable.xml --openfpga_sim_setting_file /opt/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /opt/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /opt/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /opt/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /spare/opt/OpenFPGA/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/run001/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test
INFO - Validating commnad line arguments
INFO - Run directory : /spare/opt/OpenFPGA/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/run001/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH
INFO - Runing OpenFPGA Shell Engine
INFO - Launching OpenFPGA Shell Run
INFO - OpenFPGAShell Revision: 9020577e Compiled: 2021-02-08T09:57:33
INFO - OpenFPGA Shell Run is written in file openfpgashell.log
INFO - vpr result extracted in file vpr_stat.result
INFO - Launching iverilog_verification
INFO - iverilog_verification is written in file iverilog_output.txt
INFO - Launching vvp_verification
```

tangxifan, 2021-02-09:
@nachiket I can reproduce this error on my local machine. Normally, the iverilog should not take more than 10 minutes.

nachiket, 2021-02-09:
I also patched in `dffr.v` into my XML -- since that showed up as an error during Synopsys DC synthesis. Not sure if that was needed?

nachiket, 2021-02-09:
AH! Your `dff.v` has the always block triggered on both posedge clk and set/reset.

nachiket, 2021-02-09:
Why don't I just use `dff.v`? Is there a reason I should have a separate `dffsrq.v` and `dffr.v`? It looks like you have functional code there that's quite similar to mine. Should I just point to `dff.v` in the XML instead?

nachiket, 2021-02-09:
Anyways, after adding the posedge RST/SET conditions the verification passes.
```
nachiket at jane in fpga_verilog git:(master*) 21-02-08 10:15PM python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py synthesizable_verilog
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task synthesizable_verilog
 INFO (     MainThread) - Created "run002" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - 00_and2_MIN_ROUTE_CHAN_WIDTH Finished with returncode 0, Time Taken 3 seconds
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - ***** 0 runs pending *****
 INFO (     MainThread) - Task execution completed
 ```

tangxifan, 2021-02-09:
@nachiket Glad to hear that the errors have been sorted out.

We support the use of ``dff`` actually. You can find an example [here](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/openfpga_arch/k4_N4_40nm_cc_openfpga.xml#L140)

I am not sure why ``vvp`` is painful on some Verilog codes. When we do tape-outs, we always run verification using Modelsim. You may try Modelsim and see if it works well. But before trying that, please ensure that your openfpga_shell script has turned off the ``--support_icarus_simulator``. You can see an example [here](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/openfpga_shell_scripts/mcnc_example_script.openfpga#L51).

nachiket, 2021-02-09:
Looks like I also needed `tap_buf4.v` as well, but my cbox now complies via Synopsys DC!

tangxifan, 2021-02-09:
Great. I believe we can work out a very nice tutorial from this effort!

closed 2021-02-09 by nachiket

nachiket, 2021-02-19:
If I wanted to run Modelsim instead of vvl (seems to be hanging on W>70), do we just drop this switch and rerun `run_fpga_task.py`? 

> On Feb 8, 2021, at 10:22 PM, tangxifan <notifications@github.com> wrote:
> 
> I am not sure why vvp is painful on some Verilog codes. When we do tape-outs, we always run verification using Modelsim. You may try Modelsim and see if it works well. But before trying that, please ensure that your openfpga_shell script has turned off the --support_icarus_simulator. You can see an example here <https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/openfpga_shell_scripts/mcnc_example_script.openfpga#L51>.
> 
>

tangxifan, 2021-02-19:
@nachiket We have a tutorial [here](https://openfpga.readthedocs.io/en/master/tutorials/design_flow/verilog2verification/#run-modelsim-simulation)

If it does not work, please let me know.

nachiket, 2021-02-19:
Thanks — looks like `run_modelsim.py` needs to be patched to use relative paths. 

I have a git push here: https://github.com/watcag/OpenFPGA/commit/8a026f7e4631f792301b6b988dc3ba1df37c3a56 <https://github.com/watcag/OpenFPGA/commit/8a026f7e4631f792301b6b988dc3ba1df37c3a56>

Can someone review it? I’ll submit a pull request once we get it working.

tangxifan, 2021-02-19:
No worries. Ganesh and I can review it.

nachiket, 2021-02-19:
Hmm — can I have a single task that does both Verilog generation and Modelsim simulation?

I have `synthesize_verilog` local task.
But for sim I seem to need `basic_tests/full_testbench/configuration_chain` task.conf? Is there a way to unify these into one?

tangxifan, 2021-02-19:
You can try to call the ``write_fabric_verilog`` twice, similar to the SDC writer.

tangxifan, 2021-02-19:
If you want to run Verilog generation + Modelsim flow (without any iVerilog), you can refer to this [script](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/mcnc_big20/config/task.conf)

nachiket, 2021-02-19:
I don’t quite notice any differences in the OpenFPGA scripts for config chain task vs the synth verilog tasks.
There’s an SDC being written out for configuration chain.

If I ask the `synthesizable_verilog task` to `-run_sim` I get this weird message:

COMMAND: 
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_modelsim.py synthesizable_verilog --run_sim

LOG:

INFO (MainThread) - Created log file at /blah/synthesizable_verilog/latest/modelsim_run.log
INFO (MainThread) - Found 0 INI files
INFO (MainThread) - Launching 2 parallel threads
INFO (MainThread) - = = = == = = == = = == = = == = = == = = == = = == = = == = = == = = =
INFO (MainThread) - Passed Jobs 0/0
INFO (MainThread) - Result file stored at /blah/synthesizable_verilog/latest/modelsim_result.csv
INFO (MainThread) - = = = == = = == = = == = = == = = == = = == = = == = = == = = == = = =

There’s nothing in the CSV file.

Nachiket

tangxifan, 2021-02-19:
According to my understanding, the script cannot spot the ``.ini`` file that is generated by OpenFPGA.
Note that ``write_verilog_testbench`` command can output an ``.ini`` file, which is required to run Modelsim.

https://github.com/lnis-uofu/OpenFPGA/blob/bcd8256c59b3951bbb5d7401581968aa165571d8/openfpga_flow/openfpga_shell_scripts/fix_device_route_chan_width_example_script.openfpga#L58

nachiket, 2021-02-19:
Yeah, I’m relying on the `write_verilog_testbench` command here:
$OPENFPGA_ROOT/openfpga_flow/openfpga_shell_scripts/fix_device_route_chan_width_example_script.openfpga

I deleted the icarus switch as indicated earlier.

If I run the following I get a weird error. I think the ini file should’ve been generated here.

python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_modelsim.py synthesizable_verilog

INFO (MainThread) - Created log file at /blah/synthesizable_verilog/latest/modelsim_run.log
INFO (MainThread) - Found 0 INI files
INFO (MainThread) - Created runsim and proc files
Traceback (most recent call last):
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_modelsim.py", line 321, in <module>
    main()
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_modelsim.py", line 136, in main
    results = create_tcl_script(task_ini_files)
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_modelsim.py", line 240, in create_tcl_script
    logger.info(f"runsim_filename {runsim_filename}")
UnboundLocalError: local variable 'runsim_filename' referenced before assignment

tangxifan, 2021-02-19:
@ganeshgore wrote these scripts. Can you provide insights here?

nachiket, 2021-02-19:
Please check my fork in watcag/OpenFPGA.git as I have edited the modelsim.py file to allow relative task dirs.

nachiket, 2021-02-19:
Hmm maybe I’m wrong, but the .ini file is in the `SimulationDeck/` folder, but the code checking for ini files doesn’t seem to be aware of that path.

nachiket, 2021-02-19:
Ok I’ve made some fixes to run_modelsim.py in watcag/OpenFPGA.git that seems to make it at least call vsim now.

Ini file name needs to be `simulation_deck.ini`
There *seems* to be no need for` SimulationDeck` folder name in the OpenFPGA shell scripts so I dropped that from the `write_verilgo_testbench` command.

Nachiket

nachiket, 2021-02-19:
LOL what version of Modelsim are you using? I get a Tcl error about “try” not being a valid command.

```
# invalid command name "try"
#     while executing
# "try {
#   top_create_new_project $projectname $verilog_files $project_path $simtime $unit $top_tb
# } finally {
#    quit
# }"
```

nachiket, 2021-02-19:
Another error:
```
Configuration/Project file
/uusoc/facility/cad_tools/Mentor/modelsim10.7b/modeltech/modelsim.ini
was not found.
Unable to copy settings.
```

I guess you need a default modelsim.ini file somewhere?

nachiket, 2021-02-19:
Ok looks like its atleast running with Modelsim 10.7c!

tangxifan, 2021-02-19:
@nachiket You can see the versions that are supported here

https://openfpga.readthedocs.io/en/master/tutorials/getting_started/tools/#third-party-tools

nachiket, 2021-02-19:
OK the command that eventually seems to run to completion without errors was:

```
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py synthesizable_verilog # to create RTL first!
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_modelsim.py synthesizable_verilog --run_sim --modelsim_ini /CMC/tools/mentor/modelsim.10.7c/modeltech/modelsim.ini
```

The INI path was necessary for our installed paths of Modelsim.

How do I know that this “worked”? The final message is:
```
INFO (MainThread) - = = = == = = == = = == = = == = = == = = == = = == = = == = = == = = =
INFO (MainThread) - Passed Jobs 1/1
INFO (MainThread) - Result file stored at /blah/synthesizable_verilog/latest/modelsim_result.csv
INFO (MainThread) - = = = == = = == = = == = = == = = == = = == = = == = = == = = == = = =
```

tangxifan, 2021-02-19:
Usually, you can find log files in the local run directory

nachiket, 2021-02-19:
Not sure which file we’re looking at?

Also, it turns out `run_fpga_task.py` is still running `vvp`. I did drop the icarus support from the OpenFPGA shell script. What am I missing?

ganeshgore, 2021-02-19:
@nachiket  Sorry for the delayed response I was occupied with something else. Based on my understanding of the above thread, here are my comments.
1. You can add  `--debug` flag while executing `python3.8 .... run_modelsim.py ` for more logging, you can run --help for more flags
2. Modelsim execution directly from the task setting (like iverilog) is not supported, and the preferred method is running a separate command as you shoed above
3. The ModelSim project directory is created in the same task run directory where the task script creates the ./SRC directory
4. Yes, the `modelsim.tcl` can be improved. The current version just a preliminary version for basic simulation. 

Are there any more unanswered questions? I will be happy to answer.

nachiket, 2021-02-20:
I commented out the last line:
```
#end_flow_with_test=
```

That seems to terminate things faster. Is that needed to stop a default iverilog simulation?

ganeshgore, 2021-02-20:
Yes, you are right; removing `#end_flow_with_test=` will just generate test-benches and not run any verification. You can run verification separately with `run_modelsim.py` in the following command.

nachiket, 2021-02-20:
Ok — vvp seem to hang for W>70. So that’s useful to know. Modelsim works just fine. At least the simulation completes and I don’t see any errrors. I’m still not sure how to detect if a simulation ran properly without errors.