lnis-uofu/OpenFPGA #228: Specify non-minimum channel width?

nachiket, 2021-02-09:
Is it possible to specify a non-minimum channel width in the task.conf file? Where or how might I specify this?

nachiket, 2021-02-09:
I *think* this addresses my question -- https://openfpga.readthedocs.io/en/latest/manual/openfpga_flow/run_fpga_task/?highlight=channel%20width#script-parameter-sections

tangxifan, 2021-02-09:
@nachiket Please be aware that the the current documentation about openfpga flow/task is out of date (@ganeshgore 
 already filed an issue #151 and he is working on that). This is due to the change in user-interface (a result from refactoring the code base in 2020).

This applies to customize the routing channel width. 
- There is [an example script](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/openfpga_shell_scripts/fix_device_route_chan_width_example_script.openfpga#L3). You can specify the script template to be used in the task configuration tile. Here is an example where we use the [fixed device support](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/tasks/basic_tests/fixed_device_support/config/task.conf#L19).  
- You can specify the device size and routing channel width using the variable defined in the script template. For example, the variable ``openfpga_vpr_route_chan_width``  in task configuration will replace the variable ``${OPENFPGA_VPR_ROUTE_CHAN_WIDTH}`` in the shell template.

It means that you need to create multiple task configuration file to meet your demand in sweeping routing W. We would like to study how to simplify the process.

nachiket, 2021-02-09:
Ok so I can swap in this fixed-channel width script in my task configuration file.

But, where do I specify the variable `openfpga_vpr_route_chan_width`? In the [GENERAL] portion of the `task.conf` file?

nachiket, 2021-02-09:
or do I supply is under [SYNTH_PARAM] like you have for your benchmarks individually?

tangxifan, 2021-02-09:
@nachiket You just need to specify the variable in [OPENFPGA_SHELL] section. See this [line](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/tasks/basic_tests/fixed_device_support/config/task.conf#L23)

nachiket, 2021-02-09:
Understood — so the `bench{0|1|2}_chan_width` parameters aren’t relevant  here?

tangxifan, 2021-02-09:
Yes. the ``bench{0|1|2}_chan_width`` are not considered.

nachiket, 2021-02-09:
Also -- manual sweeps seem a bit painful. I can probably sed/replace this line across my sweeps, but a command-line parameter to `run_fpga_task.py` could be useful?

tangxifan, 2021-02-09:
I bring @ganeshgore  in the loop since he is the author of these scripts.

tangxifan, 2021-02-09:
@ganeshgore Can you share your roadmap on the routing channel width?

tangxifan, 2021-02-09:
@nachiket Currently, I agree that it is painful. Sed and scripting can alleviate this pain.

nachiket, 2021-02-09:
Hmm @tangxifan I'm not able to correctly run the fixed-channel-width OpenFPGA script.

I am using `synthesizable_verilog` task. I just replaced a single line in `task.conf`
```
+ openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/fix_device_route_chan_width_example_script.openfpga
- openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/example_script.openfpga
```

I now get this error: 
```
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 442, in run_single_script
    raise subprocess.CalledProcessError(0, " ".join(command))
.... blah blah blah returned non-zero exit status 0.
```

Debugging this further:
```
python3 /opt/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /opt/OpenFPGA/openfpga_flow/vpr_arch/k6_frac_N10_tileable_40nm.xml /opt/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /spare/opt/OpenFPGA/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/run010/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /opt/OpenFPGA/openfpga_flow/openfpga_shell_scripts/fix_device_route_chan_width_example_script.openfpga --openfpga_arch_file /opt/OpenFPGA/openfpga_flow/openfpga_arch/k6_frac_N10_stdcell_mux_40nm_openfpga_synthesizable.xml --openfpga_sim_setting_file /opt/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --openfpga_vpr_device_layout 2x2 --openfpga_vpr_route_chan_width 10 --activity_file /opt/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /opt/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /opt/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /spare/opt/OpenFPGA/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/run010/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test
```

```
... blah blah blah...

ERROR - OpenFPGA Shell Run run failed with returncode 1
ERROR - command /opt/OpenFPGA/openfpga/openfpga -batch -f and2_run.openfpga
ERROR - -->>Error 1:
ERROR - -->>Error 2: Command 'vpr' execution has fatal errors
ERROR - Current working directory : /spare/opt/OpenFPGA/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/run009/k6_frac_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH
ERROR - Failed to run OpenFPGA Shell Run task
ERROR - Exiting . . . . . .```

Which further yields:
```
/opt/OpenFPGA/openfpga/openfpga -batch -f and2_run.openfpga
```

```
.... blah blah blah... 

Built a multiplexer library of 12 physical multiplexers.
Maximum multiplexer size is 64.
# Build a library of physical multiplexers took 0.00 seconds (max_rss 11.2 MiB, delta_rss +0.0 MiB)
# Build the annotation about direct connection between tiles
Built 0 tile-to-tile direct connections
# Build the annotation about direct connection between tiles took 0.00 seconds (max_rss 11.2 MiB, delta_rss +0.0 MiB)
Building annotation for mapped blocks on grid locations...Done
User specified the operating clock frequency to use VPR results
terminate called after throwing an instance of 'VprError'
  what():  in load_one_net_delay: Traceback for net 2 does not exist.

Abort (core dumped)
```

Using `example_script.openfpga` seems to work fine.

nachiket, 2021-02-09:
FYI I did add this to the conf file:
```
openfpga_vpr_device_layout=2x2
openfpga_vpr_route_chan_width=10
```

nachiket, 2021-02-09:
@ganeshgore 

For now I’m using the `synthesizable_verilog` task I created as recommended by @tangxifan in #225.
Based on recent suggestions for fixed channel width support, I’ve replaced the shell script in the `task.conf`.

[Check the other message, its not quite working yet]

I’m thinking of just doing the following ->

sed -i "s/openfpga_vpr_route_chan_width=.*/openfpga_vpr_route_chan_width=10/" synthesizable_verilog/config/task.conf

— rerun verilog generation —

sed -i "s/openfpga_vpr_route_chan_width=.*/openfpga_vpr_route_chan_width=20/" synthesizable_verilog/config/task.conf

— rerun verilog generation —

sed -i "s/openfpga_vpr_route_chan_width=.*/openfpga_vpr_route_chan_width=30/" synthesizable_verilog/config/task.conf

— rerun verilog generation —

This lets me sweep over various channel widths.

tangxifan, 2021-02-09:
@nachiket According to what I see, VPR failed due to a very small routing channel width. 
- You may have to know what is the minimum routing channel width and then specify a correct one. 
- You can always check the log file ``openfpga_shell.log`` or ``vpr_stdout.log`` for full details.

nachiket, 2021-02-09:
@tangxifan — Ah it looks like I need 20 minimum for and2. I thought its a simple enough benchmark and should work with very few channels!

ganeshgore, 2021-02-09:
@nachiket Hello Parameter sweeping is not supported from the task file directly.
In our initial experiments, we always found out the min-width (using `min_chan_width`) required and added some buffer on top of it to run further experimentation. 
However, the approach you used in the previous commit of adding multiple SCRIPT_PARAM seems reasonable. That approach runs all the combinations in parallel.

Are you looking for even fine sweeping? or something is missed in this approach?

nachiket, 2021-02-09:
To be honest I’m still exploring so I am mucking my way around the infrastructure. What would be useful if its were possible to override options in the task.conf from the run_fpga_task.py run. But, this may not be really necessary, as I’m able to conveniently run sweep experiments externally with a simple sed/replace. I’d just file this away this feature for resisting later if a real need arises for consideration once I know more about the infrastructure.

ganeshgore, 2021-02-09:
With above changes in the `task.conf`, you can change this variable `OPENFPGA_VPR_ROUTE_CHAN_WIDTH` to   `FIX_ROUTE_CHAN_WIDTH` in `.openfpgashell` file

https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/openfpga_shell_scripts/fix_device_route_chan_width_example_script.openfpga#L3

Just to give you high level idea of how task maps to run, whatever you define in the SCRIPT_PARAM section will be available in `.openfpgashell` script as variable in capital letter. and every combination `arch`, `benchmark `and `script parameter`  creates one task.

ganeshgore, 2021-02-09:
With above approach you wont have to run sed  externally

nachiket, 2021-02-09:
I'm not sure I follow. What would be the sweep methodology? I just run the python run_fpga_task.py once? And it will execute a sweep for the specified combinations? I'm not sure where I'd add the `FIX_ROUTE_CHAN_WIDTH` parameter.

ganeshgore, 2021-02-09:
So to sweep particular parameter, define the parameter in `SCRIPT_PARAM_xxxx` section in` task.conf` file, as follows

```
[SCRIPT_PARAM_Fixed_Routing_40]
fix_route_chan_width=40

[SCRIPT_PARAM_Fixed_Routing_50]
fix_route_chan_width=50

[SCRIPT_PARAM_Fixed_Routing_60]
fix_route_chan_width=60

[SCRIPT_PARAM_Fixed_Routing_70]
fix_route_chan_width=70

```
and in the`.openfpga` shell script

` vpr ${VPR_ARCH_FILE} ${VPR_TESTBENCH_BLIF} --clock_modeling route --device ${OPENFPGA_VPR_DEVICE_LAYOUT} --route_chan_width ${FIX_ROUTE_CHAN_WIDTH} `

This will run each combination of `ARCH`, `BENCHMARK` defined `SCRIPT_PARAM` as a job. so in case of this [file ](https://github.com/lnis-uofu/OpenFPGA/blob/b14b5f975da25b11d6d6f0c631d7c85e929e0451/openfpga_flow/tasks/fpga_verilog/synthesizable_verilog/config/task.conf) there will be 9 runs. First combination might fail `ARCH0 `+ `BENCH0 `+ `SCRIPT_PARAM_MIN_ROUTE_CHAN_WIDTH` as it doesn't have `fix_route_chan_width` defined

nachiket, 2021-02-09:
Understood! This was what I originally interpreted from the documentation, but good to know which variable maps where.

closed 2021-04-24 by tangxifan

narutozxp, 2023-04-12:
@ganeshgore Is it useful in the task.conf file to specify “[SCRIPT_PARAM_Slack_30] min_route_chan_width=1.3”? thanks!