lnis-uofu/OpenFPGA #230: SDC syntax for top-level FPGA file seems to have non-existing components

nachiket, 2021-02-09:
Looking at `SDC_analysis/and2_fpga_top_analysis.sdc` file, there seem to be weird top/left/right/bottom edge components that do not seem to exist in the design.

For instance, I see this error with Synopsys DC:
```
Warning: Can't find object 'sb_1__1_mux_bottom_track_17/in[1]' in design 'fpga_top'. (UID-95)
```

tangxifan, 2021-02-09:
@nachiket I think it is valid bug. I have reproduced it locally. Will patch it. On the other side, the analysis SDC is not used for backend flow. It is part of the verification where the analysis SDC is load to STA and perform timing analysis on a mapped FPGA.

nachiket, 2021-02-09:
Thanks. I have a couple of followup questions:

1. For some reason Synopsys DC does not like the top-level module name in the path. I deleted it and it accepted the constraints. Do you have if there’s a specific `read_sdc` command format that would accept the SDC? 

2. How do the specific delays for `cbox*.sdc` and `sbox*.sdc` come from? Are these extracted from RC wire and component delays from the VPR arch file? Without a clock I’m not sure I understand how to interpret the negative slack for a cbox reported by Synopsys DC. Do I look upon the max_delay constraint imposed on that path to see the actual speed of the path?

tangxifan, 2021-02-09:
@ganeshgore See if you have an idea about the ``read_sdc`` command and ``set_max_delay`` for combinational paths.

Regarding the timing constraints, it is derived from the `Rmetal``, ``Cmetal`` and ``Tdel``  in VPR architecture file.

nachiket, 2021-02-09:
Thanks.

FYI, I also had to adjust the delays from seconds to nanoseconds for some reason for Synopsys DC! Is it possible to generate the SDC files with an optional time scaling to ns? I manually adjusted the delays by multiplying them for now by 10^9.

Nachiket

tangxifan, 2021-02-09:
@nachiket We have ``--time_unit`` option available for all the SDC writer. See [documentation](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_sdc_commands/#cmdoption-time_unit)
You can update your openfpga shell script by adding these options.

nachiket, 2021-02-09:
Perfect! Thanks! Now I just need to figure out why the top-level module path in the hierarchy needs to be deleted from the path to get the SDC accepted by Synopsys DC.

tangxifan, 2021-02-09:
No worries. If it has to be removed. We can add an option to skip the top-level module in SDC files. In future, I would like to have OpenSTA in the docker image so that we can validate the SDC in CI/CD

nachiket, 2021-02-09:
One of the features that would be good to add is to remove all hierarchy for leaf-level compilation and timing analysis of individual components. I did do that manually for now, but I wonder if its too arcane a use-case for general public.

Nachiket

tangxifan, 2021-02-09:
@nachiket You may try this option ``--hierarchical``. See [documentation](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_sdc_commands/#cmdoption-hierarchical).

nachiket, 2021-02-09:
Yeah, but that means I have to rerun without hierarchical enabled for the full FPGA chip analysis. I’d rather have two sets of files — one hierarchical and one without.

tangxifan, 2021-02-09:
@nachiket  You can try running two ``write_pnr_sdc`` commands (one with hierarchical and one without) in the openfpga shell script.

nachiket, 2021-02-09:
Nice! That works splendidly.

nachiket, 2021-02-09:
A potential inconsistency in using -hierarchical when generating `logical_tile_clb_mode_clb_.sdc`. There should be a prefix `logical_tile_clb_mode_clb__0` in front of `clb_I[0] `. Internal hierarchy below a level should still be preserved, right?

tangxifan, 2021-02-09:
Indeed. We did so because @ganeshgore developed a few python scripts to post process the SDC. @ganeshgore can you share how the post-processing is done?

nachiket, 2021-02-09:
For the non-hierarchical option with `write_pnr_c` I am still seeing SDC path errors for ‘logical_tile_clb_mode_clb__0` component. The SDC only contains a single set of constraints for `grid_clb` as seen here `grid_clb/logical_tile_clb_mode_clb__0/clb_I[0]`. However, this seems to not work with `fpga_top.v` which has `grid_clb_1__1_`,… instances. Not sure if there’s a way to get these constraints replicated for all instances?

nachiket, 2021-02-09:
Even when I manually edit this, timing analysis does not seem to work on the overall FPGA chip.

I’ve added all SDC files to the Synopsys project:
```
read_sdc $env(RTLDIR)/SDC/cby_1__1_.sdc
read_sdc $env(RTLDIR)/SDC/cby_0__1_.sdc
read_sdc $env(RTLDIR)/SDC/cbx_1__0_.sdc
read_sdc $env(RTLDIR)/SDC/cbx_1__1_.sdc
read_sdc $env(RTLDIR)/SDC/sb_0__0_.sdc
read_sdc $env(RTLDIR)/SDC/sb_0__1_.sdc
read_sdc $env(RTLDIR)/SDC/sb_1__0_.sdc
read_sdc $env(RTLDIR)/SDC/sb_1__1_.sdc
read_sdc $env(RTLDIR)/SDC/logical_tile_clb_mode_clb_.sdc
read_sdc $env(RTLDIR)/SDC/logical_tile_clb_mode_default__fle_mode_physical__fabric_mode_default__ff.sdc
read_sdc $env(RTLDIR)/SDC/logical_tile_clb_mode_default__fle_mode_physical__fabric_mode_default__frac_logic.sdc
read_sdc $env(RTLDIR)/SDC/logical_tile_clb_mode_default__fle_mode_physical__fabric.sdc
read_sdc $env(RTLDIR)/SDC/logical_tile_clb_mode_default__fle.sdc
read_sdc $env(RTLDIR)/SDC/logical_tile_io_mode_io_.sdc
read_sdc $env(RTLDIR)/SDC/disable_configurable_memory_outputs.sdc
read_sdc $env(RTLDIR)/SDC/disable_configure_ports.sdc
read_sdc $env(RTLDIR)/SDC/disable_routing_multiplexer_outputs.sdc
read_sdc $env(RTLDIR)/SDC/disable_sb_outputs.sdc
read_sdc $env(RTLDIR)/SDC/global_ports.sdc
```

The timing analysis reports no constrained paths!
```
  Startpoint: grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/GPIO_0_/DIR (internal pin)
  Endpoint: gfpga_pad_GPIO_PAD[0]
            (output port)
  Path Group: (none)
  Path Type: max

  Des/Clust/Port     Wire Load Model       Library
  ------------------------------------------------
  fpga_top           5K_hvratio_1_1        NangateOpenCellLibrary

Attributes:
    d - dont_touch
    u - dont_use
   mo - map_only
   so - size_only
    i - ideal_net or ideal_network
  inf - infeasible path

  Point                                       Fanout     Trans      Incr       Path      Attributes
  ---------------------------------------------------------------------------------------------------------
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/GPIO_0_/DIR (GPIO_0)     0.00     0.00 r
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/GPIO_0_/DIR (net)     0.00     0.00 r
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/GPIO_0_/PAD_tri/Z (TBUF_X1)     0.01     0.12     0.12 f
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/GPIO_0_/PAD (net)     2     0.00     0.12 f
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/GPIO_0_/PAD (GPIO_0)     0.00     0.12 f
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/gfpga_pad_GPIO_PAD[0] (net)     0.00     0.12 f
  grid_io_top_1__2_/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/gfpga_pad_GPIO_PAD[0] (logical_tile_io_mode_physical__iopad_0)     0.00     0.12 f
  grid_io_top_1__2_/logical_tile_io_mode_io__0/gfpga_pad_GPIO_PAD[0] (net)     0.00     0.12 f
  grid_io_top_1__2_/logical_tile_io_mode_io__0/gfpga_pad_GPIO_PAD[0] (logical_tile_io_mode_io__0)     0.00     0.12 f
  grid_io_top_1__2_/gfpga_pad_GPIO_PAD[0] (net)                     0.00       0.12 f
  grid_io_top_1__2_/gfpga_pad_GPIO_PAD[0] (grid_io_top)             0.00       0.12 f
  gfpga_pad_GPIO_PAD[0] (net)                                       0.00       0.12 f
  gfpga_pad_GPIO_PAD[0] (inout)                           0.01      0.00       0.13 f
  data arrival time                                                            0.13
  ---------------------------------------------------------------------------------------------------------
  (Path is unconstrained)
```

For individual cbox, sbox, and logic block I see some numbers.

tangxifan, 2021-02-09:
@ganeshgore I believe we hit these issues before. Can you share your solution?

ganeshgore, 2021-02-10:
@nachiket Are you trying to use SDCs to synthesize modules using design compiler?

nachiket, 2021-02-10:
Yes — I’m using the SDCs for timing analysis using Design Compiler.

ganeshgore, 2021-02-10:
I never tried performing complete top level synthesis using netlist and SDCs from the OpenFPGA, 
but I suppose you should able to do that module wise.
For example, while synthesizing `cby_1__1` you can load `cby_1__1_.sdc` and `disable_configure_ports.sdc` files

nachiket, 2021-02-10:
That doesn’t seem to work for all components like CLBs as the internal hierarchy isn’t always preserved with the -hierarchical switch. I did get it to work with Cbox and Sbox components. There is also some issue with naming as reported in #229 and #230. 

For full FPGA compilation, the only thing you’d need to support is correct instance naming. But even then (after I manually fix instance names) I see a lot of timing arcs disabled warnings even with the SDC constraints for the logic block portions of the design.

ganeshgore, 2021-02-10:
Yeah, I agree. There is some work required on the SDC front, it is in our immediate TODO list. 
I will try to loop you in the discussion. Your inputs will be helpful. 

Till then we will quickly fix the paths related issues you reported and setup CI/CD environment with OpenSTA.

closed 2021-06-23 by tangxifan