lnis-uofu/OpenFPGA #232: Supply relative paths to vpr arch XML in task.conf

nachiket, 2021-02-10:
Right now I see VPR arch files in the `$OPENFPGA_ROOT/openfpga_flow/vpr_arch` folders being used in the `task.conf` file reference via `${PATH:OPENFPGA_PATH}` variable. How might I be able to supply my own XML externally and it be a relative path using $HOME variable or some other environment variable for instance? I can always hardcode the path, but in a classroom environment each student will have their own XML files. If I try to use variables like $HOME I get "bad substitution" errors.

```
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py synthesizable_verilog
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task synthesizable_verilog
 INFO (     MainThread) - Created "run010" directory for current task run
Traceback (most recent call last):
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 505, in <module>
    main()
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 99, in main
    job_run_list = generate_each_task_actions(eachtask)
  File "/opt/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 226, in generate_each_task_actions
    for _, arch_file in task_conf["ARCHITECTURES"].items():
  File "/usr/lib/python3.6/_collections_abc.py", line 744, in __iter__
    yield (key, self._mapping[key])
  File "/usr/lib/python3.6/configparser.py", line 1234, in __getitem__
    return self._parser.get(self._name, key)
  File "/usr/lib/python3.6/configparser.py", line 800, in get
    d)
  File "/usr/lib/python3.6/configparser.py", line 455, in before_get
    self._interpolate_some(parser, option, L, value, section, defaults, 1)
  File "/usr/lib/python3.6/configparser.py", line 507, in _interpolate_some
    option, section, rawval, ":".join(path)) from None
configparser.InterpolationMissingOptionError: Bad value substitution: option 'arch0' in section 'ARCHITECTURES' contains an interpolation key 'HOME' which is not a valid option name. Raw value: '${HOME}/blah/blah/blah/k6_frac_N10_tileable_40nm.xml'
```

ganeshgore, 2021-02-10:
Yes you are right, those variables are scripts internal variables not system variables so $HOME will not be replaced.
The supported variables are documented [here](https://github.com/lnis-uofu/OpenFPGA/blob/1ce94040da38215aef890d0401347758fb8945f8/openfpga_flow/scripts/run_fpga_task.py#L207). 

While creating independent task (not in repository) we expect all the files use relative path from task directory. You can do that by using `${PATH:TASK_DIR}` in the `task.conf` file, which will look like this 

```
[OpenFPGA_SHELL]
openfpga_shell_template=${PATH:TASK_DIR}/generate_fabric.openfpga
```

This variable is still not supported in the architecture files (because architecture files can be used with flow script also, it doesn't have notion of  task). However

nachiket, 2021-02-10:
Not sure if your message clipped. I was able to get the following to work:

```

[ARCHITECTURES]
arch0=${PATH:TASK_DIR}/config/k6_frac_N10_tileable_40nm.xml
```

Would this TASK_DIR approach work for all files such as benchmarks, vpr architecture files, openfpga arch files, openfpga shell scripts, etc?

nachiket, 2021-02-10:
Accidentally closed the issue!

ganeshgore, 2021-02-10:
yeah edited the message.
The documentation of all such features is lagging because of our recent restructuring and development. 
I will try to get all this in documentation asap.