lnis-uofu/OpenFPGA #258: External users have errors when building docker image on CI

tangxifan, 2021-03-01:
**Describe the bug**
Current docker image building step in CI requires the contributors to be a collaborator of the project.

**To Reproduce**
Steps to reproduce the behavior:
1. Check the error messages in the docker image building step in PR #257 and PR #247 

**Expected behavior**
The CI should not be limited to the collaborator of the project

**Screenshots**
The error message 
[image]

**Enviornment (please complete the following information):**
 - OS: ubuntu
 - Compiler gcc
 - Version: latest master

ganeshgore, 2021-03-01:
@yashton Can you look into this ?
This is the CI run 
https://github.com/lnis-uofu/OpenFPGA/actions/runs/610908414

yashton, 2021-03-01:
Will do

closed 2021-03-29 by tangxifan