lnis-uofu/OpenFPGA #268: Bitstream Format for the case of Multi Region Configuration Chain

zafarali-lm, 2021-03-12:
**Is your feature request related to a problem? Please describe.**
How the bitstream is divided  to configure using those multiple ccff_heads in case of multi_region carry chain configuration protocol? Can you tell me the logic behind the bitstream division for those ccff_heads?  

**Describe the solution you'd like**
Maybe a custom bitstream file for multi_region_config_protocol will make this less ambiguous? 

[image]

tangxifan, 2021-03-12:
@zafarali-lm OpenFPGA allows users to customize the organization of each configuration region through a [fabric key file.](https://openfpga.readthedocs.io/en/master/manual/arch_lang/fabric_key/). It means:

- OpenFPGA has a built-in fabric key generator, which cuts the configuration regions to balance the number of configuration bits in each region.
- User can write their own fabric key to cut the configuration region in the way they want.

You can always output the fabric key using the option ['-write_fabric_key'](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/setup_commands/#cmdoption-write_fabric_key)

zafarali-lm, 2021-03-16:
How can we extract out the bits of a particular region from the bitstream file?

tangxifan, 2021-03-19:
@zafarali-lm We do not have a direct support on this. It is not difficult to do. You can easily write some codes to enable this feature.  If you are interested, we can have a further discussion.

zafarali-lm, 2021-03-22:
I am interested in extracting the bitstreams for different configuration
regions. Can you please explain how we can divide the bitstreams and append
some extra bits with configuration bits to get chunks of bitstream ready to
be configured onto the FPGA fabric via the multi-region carry-chain
configuration protocol?
Thanks

tangxifan, 2021-03-23:
@zafarali-lm You can look into the Verilog testbenches (the [full testbench](https://openfpga.readthedocs.io/en/master/manual/fpga_verilog/testbench/#full-testbench)), where you can find how the bitstream is loaded to FPGA. 

When multi-region configuration chain is used, multiple bits are loaded in each programing clock cycle.
For example, you can see bitstream from 4 configuration region are loaded.
```
		prog_cycle_task(4'b0010);
		prog_cycle_task({4{1'b0}});
		prog_cycle_task(4'b0010);
		prog_cycle_task({4{1'b0}});
```
The first bit comes from the first configuration region. So on so forth.

zafarali-lm, 2021-03-24:
@tangxifan I have been already exploring that part and trying to figure out how the configuration bits from fabric_bitstream.xml can be formatted as it is formatted in the testbench. Actually I am interested in writing a script that 
- takes the fabric_bitstream.xml file
- extracts the configuration bits
- (appends extra bits if required)
- format them into a ready to configure data chunk
- to be stored in some flash memory or something.
- that could be used in active or passive configuration of the FPGA fabric.
Can we do that? Or do you have a better suggestion for doing the task?

tangxifan, 2021-03-29:
@zafarali-lm I think it is doable. The only challenge here is that the fabric_bitstream.xml file does not categorize bitstream into different regions. It is better to extract the information from the [autocheck_top_tb.v](https://openfpga.readthedocs.io/en/master/manual/fpga_verilog/testbench/#full-testbench)

closed 2021-04-10 by tangxifan

tangxifan, 2021-04-11:
@zafarali-lm I have patched the fabric bitstream writer. Hope it makes your life easier.

zafarali-lm, 2021-04-12:
Thanks, I'll check it out.