lnis-uofu/OpenFPGA #273: A complete end-to-end walkthrough

manili, 2021-03-17:
Hi all,
I'm completely getting lost through the tutorials and "full documentation". I really did not know where to request for help except here. (so I'm very sorry if this issue is not like a real issue at all.)
OK, what I want to do is creating an iCE40-like FPGA for my self, see its final Verilog netlist and finally create Verilog to SPICE GDSII outputs. Then write a simple code for this custom FPGA and compile it via OpenFPGA toolchain. So now the question is where should I start and how should I go step by step and see the results?

P.S. I think the lack of a complete and sample-based tutorial for the framework is a big problem for the newbies like me.

Best regards,
Manili

tangxifan, 2021-03-19:
@manili The task you mentioned is actually what a FPGA company should do. It requires a lot of efforts.
It means the learning curve may be very steep. You need to pick up a lot of background knowledges.
- The VTR documentation about [architecture description language](https://docs.verilogtorouting.org/en/latest/arch/) should be studied, through which you learn how to model an  FPGA architecture
- The OpenFPGA [architecture description language](https://openfpga.readthedocs.io/en/master/manual/arch_lang/) should be studied, through which you learn how to [build an FPGA fabric](https://openfpga.readthedocs.io/en/master/tutorials/design_flow/generate_fabric/)
- Afterwards, I suggest to read the tutorial about [how to verify FPGA fabric](https://openfpga.readthedocs.io/en/master/tutorials/design_flow/verilog2verification/).

tomBGU, 2021-11-06:
Hello all, 
I'm using this thread since my questions are not actual implementation issues, hope its fine.
Is the proposed workflow and provided software restricted to specific fabrication processes?
Can the final fabric be implemented within a SoC (as an eFPGA and not a stand-alone)?

Thanks in advance,
Tom

manili, 2022-04-13:
Hi all,
Seems like the VSD has a good virtual workshop around OSFPGA. I highly recommend everyone to check it out:
https://www.vlsisystemdesign.com/fpga