lnis-uofu/OpenFPGA #276: How to execute a minimal Verilog->Bitstream flow?

nmoroze, 2021-03-23:
Hi, 

I'm trying to figure out how to run a minimal flow to generate a bitstream for a given FPGA architecture using OpenFPGA. I'm currently working off the "generate_bitstream" example task, but I noticed this task does two things I don't want:

1) It performs power analysis using ace2
2) It requires a simulation settings file, even though it doesn't generate a Verilog testbench

As far as (1), I tried to disable the power analysis by setting `power_analysis` to false in the task config file. However, this gives me an error when running the flow, since the activity file is no longer generated, but still expected by the OpenFPGA shell script:
```
ERROR - -->>terminate called after throwing an instance of 'vtr::VtrError'
ERROR - -->>  what():  Error opening file and2_ace_out.act for r access: No such file or directory.
```

The other thing I tried to eliminate both (1) and (2) was to remove various lines from the OpenFPGA shell script template used by this task (`generate_bitstream_example_script.openfpga`). However, removing commands such as `link_openfpga_simulation_setting` or removing the `--activity_file` flag in `link_openfpga_arch` all cause the flow to break in some way (I can follow up with more details on what exactly I tried if that would be helpful). 

Overall, I'd really appreciate some guidance on how to adapt this flow to generate a bistream without requiring a run of ace2, and without providing a simulation settings XML file. If that's not possible, I'd also be curious as to why - as far as I an tell, power estimation is optional, and the simulation settings are only needed if you want to generate a Verilog testbench. Thanks in advance for any help!

tangxifan, 2021-03-23:
@nmoroze 
Thank you for your interests in OpenFPGA. Your advice will help us to be more user friendly.

Regarding the power analysis, you can try the following: 
- Set ``power_analysis`` to be false in task configuration file
- Remove ``--activity_file`` from your openfpga shell script
- Use a fixed clock cycle simulation setting ([example](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml)) for ``read_openfpga_simulation_setting`` command. This is required. Otherwise, OpenFPGA needs an ``.act`` file to find out the proper number of clock cycles to be applied in simulations.

You may refer to this example: 
https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/quicklogic_tests/lut_adder_test/config/task.conf#L9-L16

Regarding simulation setting file, I think your comment is valid. Currently, it remains a required file to run any design flows. But it has no impact on bitstream-generation-only flows. So, the hard requirements should be removed. We will see how to patch OpenFPGA. For your current use, I suggest to use an example simulation setting file, which should not impact your bitstream generation.

Feel free to comment.

nmoroze, 2021-03-23:
Thank you very much for the prompt and helpful reply! I was able to do just as you said, remove the power analysis and instead use fixed clock cycle simulation settings. 

I do think it would be nice to remove the hard requirement of providing a simulation settings file, but at the moment this is sufficient for my purposes. Feel free to close the issue, unless you wanted to use it for tracking a future patch to remove the sim settings requirement.

tangxifan, 2021-03-23:
@nmoroze No worries. I will keep this issue open as a patch is required

nmoroze, 2021-03-24:
As a quick follow-up question: would it be theoretically possible to generate the bitstream without requiring the OpenFPGA architecture file (i.e. using only the VPR architecture file)? Or does the OpenFPGA architecture file contain information necessary for bitstream generation? Thanks!

tangxifan, 2021-03-24:
@nmoroze  Unfortunately, the openfpga architecture file is required because bitstream generation is closely related to the circuit implementation of primitive modules as well as the configuration protocol, while VPR focus on the logical representation of FPGA architectures.

tangxifan, 2022-06-02:
Close this issue. If you have any follow-up questions, feel free to post in discussion forum or create an issue.

closed 2022-06-02 by tangxifan