lnis-uofu/OpenFPGA #280: ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH

tonykoz, 2021-03-29:
I am running Ubuntu 18.04 and installed OpenFPGA and compiled it. When I run the following command to confirm that I compiled correctly, I get the following error:

python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification --debug --show_thread_logs

I get the following error:

ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH

[image]

tangxifan, 2021-03-29:
@tonykoz You can see that the error is actually one line before your red rectangle. It said that the python module ``envyaml`` has not been installed. If you use pip to install the package, the issue should be solved.

tonykoz, 2021-03-29:
Thanks. That fixed it.

closed 2021-03-29 by tangxifan