lnis-uofu/OpenFPGA #288: Error in Linking Phase when compiling OpenFPGA files.

ishmamiqbal, 2021-04-14:
**Describe the bug**
Hello, 
I am running into the following error when trying to compile OpenFPGA for the first time (also attached in the screenshot):

In file included from /home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_default_converter.hpp:8:0,

                 from /home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.hpp:12,

                 from /home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp:8:

/home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp: In constructor ‘constexpr argparse::ArgValue<bool>::ArgValue()’:

/home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp:44:11: error: conversion from ‘const char [1]’ to non-scalar type ‘std::string {aka std::basic_string<char>}’ requested

     class ArgValue {
           ^
/home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp:44:11: error: conversion from ‘const char [1]’ to non-scalar type ‘std::string {aka std::basic_string<char>}’ requested

/home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp: In constructor ‘argparse::ArgumentParser::ArgumentParser(std::string, std::string, std::ostream&)’:

/home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp:21:17: note: synthesized method ‘constexpr argparse::ArgValue<bool>::ArgValue()’ first required here 

         , os_(os)
                 ^
**To Reproduce**
Steps to reproduce the behavior:

cd OpenFPGA
make all

Error occurs during linking phase

**Expected behavior**
Since I seem to have all the required packages, the compilation should have completed without any errors.

**Screenshots**
[image]

**Enviornment (please complete the following information):**
 - OS: CentOS 7
 - Compiler: gcc
 - Version: https://github.com/lnis-uofu/OpenFPGA.git 

**Additional context**
I am running centOS on a remote desktop.

tangxifan, 2021-04-14:
@ishmamiqbal This is mostly due to the use of  incompatible gcc version. The dependency of OpenFPGA is available in online [documentation](https://openfpga.readthedocs.io/en/master/tutorials/getting_started/compile/#general-guidelines), where the gcc version should be at least gcc-5.  Also in our online tests, you can see that these gcc compilers are continuously under testing. For example, https://github.com/lnis-uofu/OpenFPGA/actions/runs/748359473

ishmamiqbal, 2021-04-14:
Hello, 
I checked my gcc version, and it seems to be version 8.3.0 
Commands:

$ gcc --version
gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I also made sure to use "module load gcc" before compilation on my 2nd try, but I still got the same error. What else should I try? 

Thank you for the help!

tangxifan, 2021-04-14:
@ishmamiqbal Can you check the compilation log files? When the compilation starts, it will echo the gcc version which is actually used. It matters.

ishmamiqbal, 2021-04-18:
Hello, my terminal simply shows 'gcc' during compiling, it does not specify the version. How can I view what gcc version is being used during compilation?

tangxifan, 2021-04-18:
@ishmamiqbal Would you mind share your log files about compilation? I can take a look. Thanks

ishmamiqbal, 2021-04-19:
Hi Dr. Tang, I sent you an email with the log files (I had trouble uploading it directly to Github, my apologies!). Please let me know if this was what you were asking for, or if I was supposed to send something else.

tangxifan, 2021-04-19:
@ishmamiqbal No worries. The log file does not show the GCC version because it is based on the previous compilation. So the echo lines are reduced.
Can you try to clean up the compilation and recompile?
Something like:

```
make clean
make all
```

ishmamiqbal, 2021-04-20:
After cleaning up and recompiling, I got:

$ make all
git submodule init
git submodule update --init --recursive
mkdir -p build && cd build && cmake3  ..
-- CMAKE_BUILD_TYPE: Release
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 8.3.0

It looks like my C compiler is running 4.8.5 instead of 8.3.0. How can I make it run the updated version? The only modifications I made to my cmake file was that I replaced 
"CMAKE_COMMAND := cmake" to "CMAKE_COMMAND := cmake3"
Could that be causing this problem?

tangxifan, 2021-04-20:
@ishmamiqbal Regarding environment setup, I suggest to ask server managers for help.
Using cmake, you can specific which C/C++ compiler to use through options ``-DCMAKE_C_COMPILER=gcc`` and ``-DCMAKE_CXX_COMPILER=g++`` 

Alternatively, you can try to set environmental variables ``CC`` and ``CXX`` in your shell.

ishmamiqbal, 2021-04-20:
I managed to set gcc to version 8.3.0 during compilation, using:

module load gcc
export CC=gcc 

However, now I am getting the following error message:
In file included from /usr/include/tbb/enumerable_thread_specific.h:32,
                 from /usr/include/tbb/combinable.h:32,
                 from /home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libtatum/libtatum/tatum/graph_walkers/ParallelLevelizedWalker.hpp:8,
                 from /home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libtatum/libtatum/tatum/graph_walkers.hpp:9,
                 from /home/shay/a/iqbal11/OpenFPGA/libs/EXTERNAL/libtatum/libtatum/tatum/analyzer_factory.hpp:10,
                 from /home/shay/a/iqbal11/OpenFPGA/vpr/src/timing/timing_info.h:6,
                 from /home/shay/a/iqbal11/OpenFPGA/vpr/src/analysis/timing_reports.cpp:10:
/usr/include/tbb/concurrent_vector.h: In member function ‘tbb::concurrent_vector<T, A>::iterator tbb::concurrent_vector<T, A>::grow_by(tbb::concurrent_vector<T, A>::size_type)’:
/usr/include/tbb/concurrent_vector.h:667:38: error: operands to ?: have different types ‘tbb::internal::concurrent_vector_base_v3::size_type’ {aka ‘long unsigned int’} and ‘tbb::atomic<long unsigned int>’
         return iterator(*this, delta ? internal_grow_by( delta, sizeof(T), &initialize_array, NULL ) : my_early_size);
                                ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/tbb/concurrent_vector.h:667:38: note:   and each type can be converted to the other
/usr/include/tbb/concurrent_vector.h: In member function ‘tbb::concurrent_vector<T, A>::iterator tbb::concurrent_vector<T, A>::grow_by(tbb::concurrent_vector<T, A>::size_type, tbb::concurrent_vector<T, A>::const_reference)’:
/usr/include/tbb/concurrent_vector.h:680:38: error: operands to ?: have different types ‘tbb::internal::concurrent_vector_base_v3::size_type’ {aka ‘long unsigned int’} and ‘tbb::atomic<long unsigned int>’
         return iterator(*this, delta ? internal_grow_by( delta, sizeof(T), &initialize_array_by, static_cast<const void*>(&t) ) : my_early_size);
                                ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/tbb/concurrent_vector.h:680:38: note:   and each type can be converted to the other
make[3]: *** [vpr/CMakeFiles/libvpr.dir/src/analysis/timing_reports.cpp.o] Error 1
make[3]: Leaving directory `/home/shay/a/iqbal11/OpenFPGA/build'
make[2]: *** [vpr/CMakeFiles/libvpr.dir/all] Error 2
make[2]: Leaving directory `/home/shay/a/iqbal11/OpenFPGA/build'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/shay/a/iqbal11/OpenFPGA/build'
make: *** [all] Error 2

I am not really sure what this means. What else should I fix?

tangxifan, 2021-04-20:
@ishmamiqbal I think it is due to the use TBB workers. I usually turn the TBB support off when compilation.  Therefore, the errors are not familiar to me. I suggest you either turn it off or ask VTR project developers for help: https://github.com/verilog-to-routing/vtr-verilog-to-routing

ishmamiqbal, 2021-04-21:
How were you able to switch off TBB? I tried to edit my Makefile by adding a "-D WITH_TBB=OFF" in my "make all" rule, but I am still getting the same error.

tangxifan, 2021-04-22:
@ishmamiqbal I have not tried TBB before. TBB compilation is defined as part of the vpr CMakefile. You can try to turn it off there: 
https://github.com/lnis-uofu/OpenFPGA/blob/c1982733789261f1fd3f5eae0271046c7c68b9e1/vpr/CMakeLists.txt#L179-217

ishmamiqbal, 2021-04-28:
I managed to finish my compilation using the following commands:

ml gcc/6.3.0
ml intel
source /package/intel/18.0.1/bin/compilervars.sh intel64
export CC=icc
export CXX=icpc
cd OpenFPGA
make -j 12

However, when I check my compilation using the verification command, I get the following error message:

$ python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification --debug --show_thread_logs
 INFO (     MainThread) - Setting loggger in debug mode
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification
 INFO (     MainThread) - Created "run006" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
DEBUG (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running OpenFPGA flow with [['python3', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif', '--top_module', 'and2', '--run_dir', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/tasks/compilation_verification/run006/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH', '--fpga_flow', 'vpr_blif', '--openfpga_shell_template', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga', '--openfpga_arch_file', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml', '--openfpga_sim_setting_file', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml', '--activity_file', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act', '--base_verilog', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v', '--power', '--power_tech', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml', '--vpr_fpga_verilog', '--vpr_fpga_verilog_dir', '/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/tasks/compilation_verification/run006/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH', '--vpr_fpga_x2p_rename_illegal_port', '--end_flow_with_test', '--vpr_fpga_verilog_formal_verification_top_netlist', '--debug']]
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Traceback (most recent call last):
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/shay/a/iqbal11/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 22, in <module>
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     from envyaml import EnvYAML
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - ModuleNotFoundError: No module named 'envyaml'
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 451, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/tasks/compilation_verification/run006/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/shay/a/iqbal11/OpenFPGA/openfpga_flow/tasks/compilation_verification/run006/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist --debug' returned non-zero exit status 0.

What could be causing this?

tangxifan, 2021-04-28:
@ishmamiqbal This is due to the missing python packages. You can use pip to install the dependencies as listed in https://github.com/lnis-uofu/OpenFPGA/blob/master/requirements.txt

closed 2021-06-10 by tangxifan