lnis-uofu/OpenFPGA #313: error while Quick Compilation Verification

msaideroglu, 2021-05-21:
When I finished make all command of OpenFPGA, it successfully finishes it seems. I have already installed all dependencies seems on the page. However it gives error during first verification. 
the error script is: 
" python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification --debug --show_thread_logs
 INFO (     MainThread) - Setting loggger in debug mode
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification
 INFO (     MainThread) - Created "run002" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
DEBUG (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running OpenFPGA flow with [['python3', '/home/msaid/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py', '/home/msaid/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml', '/home/msaid/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif', '--top_module', 'and2', '--run_dir', '/home/msaid/OpenFPGA/openfpga_flow/tasks/compilation_verification/run002/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH', '--fpga_flow', 'vpr_blif', '--openfpga_shell_template', '/home/msaid/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga', '--openfpga_arch_file', '/home/msaid/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml', '--openfpga_sim_setting_file', '/home/msaid/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml', '--activity_file', '/home/msaid/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act', '--base_verilog', '/home/msaid/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v', '--power', '--power_tech', '/home/msaid/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml', '--vpr_fpga_verilog', '--vpr_fpga_verilog_dir', '/home/msaid/OpenFPGA/openfpga_flow/tasks/compilation_verification/run002/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH', '--vpr_fpga_x2p_rename_illegal_port', '--end_flow_with_test', '--vpr_fpga_verilog_formal_verification_top_netlist', '--debug']]
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Traceback (most recent call last):
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/msaid/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 22, in <module>
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     from envyaml import EnvYAML
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - ModuleNotFoundError: No module named 'envyaml'
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 451, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/msaid/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/msaid/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/msaid/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /home/msaid/OpenFPGA/openfpga_flow/tasks/compilation_verification/run002/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/msaid/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/msaid/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/msaid/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /home/msaid/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /home/msaid/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /home/msaid/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/msaid/OpenFPGA/openfpga_flow/tasks/compilation_verification/run002/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist --debug' returned non-zero exit status 0.
"

tangxifan, 2021-05-21:
@msaideroglu Please find details in the closed issue #280

msaideroglu, 2021-05-21:
Thanks it is solved.

closed 2021-05-21 by msaideroglu