lnis-uofu/OpenFPGA #315: Dependencies not found

Fre29, 2021-05-27:
Which part of OpenFPGA is buggy:
--- >Documentation

Hi,
I'm trying to install OpenFPGA on my laptop (with Ubuntu 21.04) following all steps mentioned in the tutorial "OpenFPGA tutorial: How to install OpenFPGA?". Unfortunately when it's time to verify if the tool is well compiled, with the command: 

python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification --debug --show_thread_logs

I receive these errors:

[image]

I think the problem is linked with installation of the dependencies as I needed to change some of them. More specifically, when I attempt to install those recommended in this link:

https://github.com/lnis-uofu/OpenFPGA/blob/master/.github/workflows/install_dependencies_build.sh

I get this message:

[image]

So I found these alternative possibilities: 

- universal-ctags instead of ctags

- qtbase5-dev instead of qt5-default

Are these solutions wrong?

Enviornment:
- OS: 
Ubuntu 21.04 

- Compiler:
gcc-9 ---> 9.3.0

TOM-once, 2021-06-01:
Your installation is missing EnvYAML
You need to
python -m pip install EnvYAML
pip install --upgrade EnvYAML

tangxifan, 2021-06-01:
@Fre29 Please find details in #280 

@TOM-once Thanks for the help! That is the solution.

Fre29, 2021-06-01:
Thanks, both of you. Now it's working correctly!

closed 2021-06-01 by Fre29