lnis-uofu/OpenFPGA #318: Cannot find the model for subcircuit

Chloe123Jackson, 2021-05-29:
I have been trying to create a .act file for a .blif file of my own, but it shows the following error:

Line 10: Cannot find the model for subcircuit NOT.
Reading network from file has failed.
/home/user/OpenFPGA/ace2/SRC/ace.c:357 main: Assertion 'ntk' failed.

I'm not sure how to fix it. It would be helpful to get an insight on the same.

tangxifan, 2021-05-29:
Hi @Chloe123Jackson , there could be various reasons why ACE failed. Would you mind sharing your .blif file?
According to what you say ``Cannot find the model for subcircuit NOT``, it seems that your BLIF file contains subcircuit ``.model`` other than the LUTs ``.names`` and FFs ``.latch``.

Chloe123Jackson, 2021-05-31:
Yeah sure. The blif file has been provided below:

# Generated by Yosys 0.9+2406 (git sha1 f44a4f90, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)

.model Slice
.inputs FXINA FXINB BY Clk CE BX Clr FLUT_sel[0] FLUT_sel[1] FLUT_sel[2] FLUT_sel[3] GLUT_sel[0] GLUT_sel[1] GLUT_sel[2] GLUT_sel[3] FLUT_ram[0] FLUT_ram[1] FLUT_ram[2] FLUT_ram[3] FLUT_ram[4] FLUT_ram[5] FLUT_ram[6] FLUT_ram[7] FLUT_ram[8] FLUT_ram[9] FLUT_ram[10] FLUT_ram[11] FLUT_ram[12] FLUT_ram[13] FLUT_ram[14] FLUT_ram[15] GLUT_ram[0] GLUT_ram[1] GLUT_ram[2] GLUT_ram[3] GLUT_ram[4] GLUT_ram[5] GLUT_ram[6] GLUT_ram[7] GLUT_ram[8] GLUT_ram[9] GLUT_ram[10] GLUT_ram[11] GLUT_ram[12] GLUT_ram[13] GLUT_ram[14] GLUT_ram[15] FLUT_we GLUT_we GYMUX_sel DYMUX_sel FXMUX_sel DXMUX_sel
.outputs Fi Y YQ F5 X XQ
.names $false
.names $true
1
.names $undef
.subckt NOT A=GYMUX_sel Y=$abc$151$new_n21_
.subckt NOT A=DYMUX_sel Y=$abc$151$new_n22_
.subckt NOT A=BX Y=$abc$151$new_n23_
.subckt NOT A=FXMUX_sel Y=$abc$151$new_n24_
.subckt NOT A=DXMUX_sel Y=$abc$151$new_n25_
.subckt NOT A=FXINB Y=$abc$151$new_n26_
.subckt NOT A=GLUT_out Y=$abc$151$new_n27_
.subckt NOT A=FLUT_out Y=$abc$151$new_n28_
.subckt NAND A=BY B=FXINA Y=$abc$151$new_n29_
.subckt NOT A=$abc$151$new_n29_ Y=$abc$151$new_n30_
.subckt NOR A=BY B=$abc$151$new_n26_ Y=$abc$151$new_n31_
.subckt NOR A=$abc$151$new_n30_ B=$abc$151$new_n31_ Y=$abc$151$new_n32_
.subckt NOT A=$abc$151$new_n32_ Y=Fi
.subckt NOR A=$abc$151$new_n24_ B=$abc$151$new_n32_ Y=$abc$151$new_n34_
.subckt NOR A=FXMUX_sel B=$abc$151$new_n28_ Y=$abc$151$new_n35_
.subckt NOR A=$abc$151$new_n34_ B=$abc$151$new_n35_ Y=$abc$151$new_n36_
.subckt NOT A=$abc$151$new_n36_ Y=X
.subckt NOR A=$abc$151$new_n25_ B=$abc$151$new_n36_ Y=$abc$151$new_n38_
.subckt NAND A=BX B=$abc$151$new_n25_ Y=$abc$151$new_n39_
.subckt NAND A=CE B=$abc$151$new_n39_ Y=$abc$151$new_n40_
.subckt NOR A=$abc$151$new_n38_ B=$abc$151$new_n40_ Y=$abc$151$new_n41_
.subckt NOR A=CE B=XQ Y=$abc$151$new_n42_
.subckt NOT A=$abc$151$new_n42_ Y=$abc$151$new_n43_
.subckt NAND A=Clr B=$abc$151$new_n43_ Y=$abc$151$new_n44_
.subckt NOR A=$abc$151$new_n41_ B=$abc$151$new_n44_ Y=$0\XQ[0:0]
.subckt NOR A=$abc$151$new_n21_ B=$abc$151$new_n32_ Y=$abc$151$new_n46_
.subckt NOR A=GYMUX_sel B=$abc$151$new_n27_ Y=$abc$151$new_n47_
.subckt NOR A=$abc$151$new_n46_ B=$abc$151$new_n47_ Y=$abc$151$new_n48_
.subckt NOT A=$abc$151$new_n48_ Y=Y
.subckt NOR A=$abc$151$new_n22_ B=$abc$151$new_n48_ Y=$abc$151$new_n50_
.subckt NAND A=BY B=$abc$151$new_n22_ Y=$abc$151$new_n51_
.subckt NAND A=CE B=$abc$151$new_n51_ Y=$abc$151$new_n52_
.subckt NOR A=$abc$151$new_n50_ B=$abc$151$new_n52_ Y=$abc$151$new_n53_
.subckt NOR A=YQ B=CE Y=$abc$151$new_n54_
.subckt NOT A=$abc$151$new_n54_ Y=$abc$151$new_n55_
.subckt NAND A=Clr B=$abc$151$new_n55_ Y=$abc$151$new_n56_
.subckt NOR A=$abc$151$new_n53_ B=$abc$151$new_n56_ Y=$0\YQ[0:0]
.subckt NAND A=BX B=GLUT_out Y=$abc$151$new_n58_
.subckt NAND A=$abc$151$new_n23_ B=FLUT_out Y=$abc$151$new_n59_
.subckt NAND A=$abc$151$new_n58_ B=$abc$151$new_n59_ Y=F5
.subckt DFF C=Clk D=$0\XQ[0:0] Q=XQ
.subckt DFF C=Clk D=$0\YQ[0:0] Q=YQ
.subckt lut_4 clk=Clk lut_out=FLUT_out lut_sel[0]=FLUT_sel[0] lut_sel[1]=FLUT_sel[1] lut_sel[2]=FLUT_sel[2] lut_sel[3]=FLUT_sel[3] ram_i[0]=FLUT_ram[0] ram_i[1]=FLUT_ram[1] ram_i[2]=FLUT_ram[2] ram_i[3]=FLUT_ram[3] ram_i[4]=FLUT_ram[4] ram_i[5]=FLUT_ram[5] ram_i[6]=FLUT_ram[6] ram_i[7]=FLUT_ram[7] ram_i[8]=FLUT_ram[8] ram_i[9]=FLUT_ram[9] ram_i[10]=FLUT_ram[10] ram_i[11]=FLUT_ram[11] ram_i[12]=FLUT_ram[12] ram_i[13]=FLUT_ram[13] ram_i[14]=FLUT_ram[14] ram_i[15]=FLUT_ram[15] we=FLUT_we
.subckt lut_4 clk=Clk lut_out=GLUT_out lut_sel[0]=GLUT_sel[0] lut_sel[1]=GLUT_sel[1] lut_sel[2]=GLUT_sel[2] lut_sel[3]=GLUT_sel[3] ram_i[0]=GLUT_ram[0] ram_i[1]=GLUT_ram[1] ram_i[2]=GLUT_ram[2] ram_i[3]=GLUT_ram[3] ram_i[4]=GLUT_ram[4] ram_i[5]=GLUT_ram[5] ram_i[6]=GLUT_ram[6] ram_i[7]=GLUT_ram[7] ram_i[8]=GLUT_ram[8] ram_i[9]=GLUT_ram[9] ram_i[10]=GLUT_ram[10] ram_i[11]=GLUT_ram[11] ram_i[12]=GLUT_ram[12] ram_i[13]=GLUT_ram[13] ram_i[14]=GLUT_ram[14] ram_i[15]=GLUT_ram[15] we=GLUT_we
.end

tangxifan, 2021-05-31:
@Chloe123Jackson 
This is because in the ``.blif`` file, the model of ``NOT``, ``NAND``, ``DFF``, lut_4`` are not defined.
You need to define them at the end of the ``.blif`` file.
For example,

```
.model DFF
.inputs C D
.outputs Q
```

Chloe123Jackson, 2021-06-02:
@tangxifan  Thank you for the help. This error doesn't occur now

tangxifan, 2021-06-02:
@Chloe123Jackson No worries. Feel free to contact whenever you see an issue. If you do not mind, I will close this issue.

closed 2021-06-04 by tangxifan