lnis-uofu/OpenFPGA #319: Rework OpenFPGA's Verilog testbench generator

tangxifan, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
Currently, OpenFPGA's Verilog testbench generator has a major problem
- The bitstream is hardcoded in the full testbenches which causes a large file size. 
- This results in that the bitstream files outputted by OpenFPGA is not verified (it is never used as an input of a Verilog testbench)

For example: 
```
// ----- Begin bitstream loading during configuration phase -----
initial
	begin
// ----- Configuration chain default input -----
		ccff_head[0] = 1'b0;
		prog_cycle_task(1'b1);
		prog_cycle_task(1'b1);
		prog_cycle_task(1'b0);
		prog_cycle_task(1'b0);
		prog_cycle_task(1'b0);
```

**Describe the solution you'd like**
We can use the ``readmemb`` command in Verilog to read the bitstream file in Verilog testbench. And then load the bitstream using a for loop.

For example,

```
$readmemb("top_bitstream.bit", configBit);

// Program Bitstream
for(ii=0; ii&lt;(BITSTREAM_LENGTH+1); ii=ii+1) begin
   #1 prog_clk_pad <= 0;
   ccff_head_pad &lt;= configBit[ii];
   #1 prog_clk_pad &lt;= 1;
    if (ii%1000 == 0)       
      $display("Programmed %d", ii);
end
```

ganeshgore, 2021-06-02:
You can use this for switching between full and partial configuration

```
$readmemb("top_bitstream.bit", configBit);
# 0 if partial config mode 1 if full config mode 
start_config = x 

// Program Bitstream
for(ii=0; ii <(BITSTREAM_LENGTH+1); ii=ii+1) begin
   start_config = start_config | configBit[ii];

   if (start_config == 1'b1) begin
    #1 prog_clk_pad <= 0;
    ccff_head_pad &lt;= configBit[ii];
    #1 prog_clk_pad &lt;= 1;
        if (ii%1000 == 0)       
        $display("Programmed %d", ii);
   end 
end
```

ganeshgore, 2021-06-02:
This will be more modular 

```
$readmemb("top_bitstream.bit", configBit);

# 0 if partial config mode 1 if full config mode  
assign bitloc=0;
assign ccff_head = mem[bitloc]

always 
    #2 prog_Clk =~ prog_clk

always (posedge progclk)
    bitloc =bitloc+1

initial begin  

    // Program Bitstream
    for(ii=0; ii <(BITSTREAM_LENGTH+1); ii=ii+1) begin
        if mem[ii] ==1:
            bitloc = i;
    end
end
```

closed 2021-06-10 by tangxifan

tangxifan, 2021-06-15:
Final version: (take the example of a single-head configuration chain)

```
// ----- Begin bitstream loading during configuration phase -----
`define BITSTREAM_LENGTH 527
`define BITSTREAM_WIDTH 1
// ----- Virtual memory to store the bitstream from external file -----
reg [0:`BITSTREAM_WIDTH - 1] bit_mem[0:`BITSTREAM_LENGTH - 1];
reg [$clog2(`BITSTREAM_LENGTH):0] bit_index;
// ----- Registers used for fast configuration logic -----
reg [$clog2(`BITSTREAM_LENGTH):0] ibit;
reg [0:0] skip_bits;
// ----- Preload bitstream file to a virtual memory -----
initial begin
	$readmemb("fabric_bitstream.bit", bit_mem);
// ----- Configuration chain default input -----
	ccff_head[0] <= 1'b0;
	bit_index <= 0;
	skip_bits[0] <= 1'b0;
	for (ibit = 0; ibit < `BITSTREAM_LENGTH + 1; ibit = ibit + 1) begin
		if (1'b0 == bit_mem[ibit]) begin
			if (1'b1 == skip_bits[0]) begin
				bit_index <= bit_index + 1;
			end
		end else begin
			skip_bits[0] <= 1'b0;
		end
	end
end
always @(negedge prog_clock_reg[0]) begin
	if (bit_index >= `BITSTREAM_LENGTH) begin
		config_done[0] <= 1'b1;
	end else begin
		ccff_head[0] <= bit_mem[bit_index];
		bit_index <= bit_index + 1;
	end
end
```

camilaqd, 2025-03-20:
Could someone please explain the section that uses skip_bits[0]? i don' quite understand why this section is important, especially since skip_bits[0] is never assigned to 1'b1.

tangxifan, 2025-03-20:
@camilaqd This is a good question. Thank you for your interests in OpenFPGA. 
The flag is used to perform fast configuration in Verilog testbench.
See the detailed explanation in the [documentation](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-fast_configuration) 
The ``skip_bits`` is enabled only when this option is enabled.

- [Source Code](https://github.com/lnis-uofu/OpenFPGA/blob/8ef72094bf19735c269f81f818e703052caeea97/openfpga/src/fpga_verilog/verilog_top_testbench.cpp#L1800-L1801) 

- [An example in regression test](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/basic_tests/full_testbench/fast_configuration_chain/config/task.conf)

This feature is here to ensure that we do not need to change bitstream to simulate the fast configuration case. The Verilog testbench shows an example how the firmware should work.

camilaqd, 2025-03-26:
I understand. Thank you for your answer!