lnis-uofu/OpenFPGA #329: Restricting runtime memory usage

abdullahyildiz, 2021-06-15:
Is there any option to limit memory usage of the tools within OpenFPGA flow?

We get the following error when we try to run yosys_vpr flow for a custom design:

[image]

It seems that iverilog consumes all available memory space and causes the process to be killed.

tangxifan, 2021-06-16:
@abdullahyildiz I have not seen such problem before. It may be due to that I always use Modelsim to verify large designs. 
Would you mind provide some statistics which can help spot the problem?
 - What is number of LUTs or other resources used in this custom design?
 - What is the file size of the Verilog testbenches?
 - What is the memory size of your computer?

Thanks.

abdullahyildiz, 2021-06-16:
@tangxifan,

Actually the custom design is not quite big and the problem arises during the execution of the following command:

`iverilog -o compiled_cordic ./SRC/cordic_include_netlists.v -s cordic_autocheck_top_tb`

- There are 1305 4-LUTs according to vpr log.
- The size of the Verilog testbenches are around 167MB and 186MB, respectively.
- Memory size of the computer is around 8GB.

I've also added the respective run folder for reference.

[cordic.zip](https://github.com/lnis-uofu/OpenFPGA/files/6664853/cordic.zip)

tangxifan, 2021-06-17:
@abdullahyildiz I did try your codes on my side. iVerilog reports some errors but not memory explosion.

I am using ``iVerilog v10.3``

```
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_0_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_1_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_2_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_3_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_4_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_5_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_6_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_7_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_8_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_9_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_10_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_11_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_12_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_13_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_14_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_15_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_i_16_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_0_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_1_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_2_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_3_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_4_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_5_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_6_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_7_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_8_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_9_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_10_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_11_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_12_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_13_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_14_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_15_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``y_i_16_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_0_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_1_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_2_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_3_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_4_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_5_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_6_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_7_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_8_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_9_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_10_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_11_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_12_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_13_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_14_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_15_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_i_16_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_0_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_1_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_2_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_3_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_4_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_5_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_6_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``x_o_7_'' is not a port of REF_DUT.
[... cut ...]
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_o_12_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_o_13_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_o_14_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_o_15_'' is not a port of REF_DUT.
././SRC/cordic_autocheck_top_tb.v:1170: error: port ``theta_o_16_'' is not a port of REF_DUT.
102 error(s) during elaboration.
```

abdullahyildiz, 2021-06-17:
I got the same errors as above when I increase the memory size.

It seems that port mappings in the testbench are not correct. Port names in the synthesized netlist are different than what the testbench expects:

```
cordic(clk, rst, \x_i[0] , \x_i[1] , \x_i[2] , \x_i[3] , \x_i[4] , \x_i[5] , \x_i[6] , \x_i[7] , \x_i[8] , \x_i[9] , \x_i[10] , \x_i[11] , \x_i[12] , \x_i[13] , \x_i[14] , \x_i[15] , \x_i[16] , \y_i[0] , \y_i[1] , \y_i[2] , \y_i[3] , \y_i[4] , \y_i[5] , \y_i[6] , \
\y_i[7] , \y_i[8] , \y_i[9] , \y_i[10] , \y_i[11] , \y_i[12] , \y_i[13] , \y_i[14] , \y_i[15] , \y_i[16] , \theta_i[0] , \theta_i[1] , \theta_i[2] , \theta_i[3] , \theta_i[4] , \theta_i[5] , \theta_i[6] , \theta_i[7] , \theta_i[8] , \theta_i[9] , \theta_i[10] , \theta_i[\
11] , \theta_i[12] , \theta_i[13] , \theta_i[14] , \theta_i[15] , \theta_i[16] , \x_o[0] , \x_o[1] , \x_o[2] , \x_o[3] , \x_o[4] , \x_o[5] , \x_o[6] , \x_o[7] , \x_o[8] , \x_o[9] , \x_o[10] , \x_o[11] , \x_o[12] , \x_o[13] , \x_o[14] , \x_o[15] , \x_o[16] , \y_o[0] , \y_\
o[1] , \y_o[2] , \y_o[3] , \y_o[4] , \y_o[5] , \y_o[6] , \y_o[7] , \y_o[8] , \y_o[9] , \y_o[10] , \y_o[11] , \y_o[12] , \y_o[13] , \y_o[14] , \y_o[15] , \y_o[16] , \theta_o[0] , \theta_o[1] , \theta_o[2] , \theta_o[3] , \theta_o[4] , \theta_o[5] , \theta_o[6] , \theta_o[\
7] , \theta_o[8] , \theta_o[9] , \theta_o[10] , \theta_o[11] , \theta_o[12] , \theta_o[13] , \theta_o[14] , \theta_o[15] , \theta_o[16] );
```

tangxifan, 2021-06-18:
@abdullahyildiz  You are right. This is a known problem. Currently, our solution is to use implicit port mapping when call testbench generator. 
For example: 

https://github.com/lnis-uofu/OpenFPGA/blob/b16de387f92d132ccc0d9b682aab1c64e15e39ea/openfpga_flow/openfpga_shell_scripts/implicit_verilog_example_script.openfpga#L58-L60

abdullahyildiz, 2021-06-19:
@tangxifan,

Thanks for the explanation.

closed 2021-06-19 by abdullahyildiz