lnis-uofu/OpenFPGA #330: curr_task_dir is not defined

lukefahr, 2021-06-16:
https://github.com/lnis-uofu/OpenFPGA/blob/36113d35ac38e9ce10cec45dfbd18cac9e0c12de/openfpga_flow/scripts/run_fpga_task.py#L177

Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 514, in <module>
    main()
  File "openfpga_flow/scripts/run_fpga_task.py", line 99, in main
    job_run_list = generate_each_task_actions(eachtask)
  File "openfpga_flow/scripts/run_fpga_task.py", line 177, in generate_each_task_actions
    clean_up_and_exit("Task directory [%s] not found" % curr_task_dir)
UnboundLocalError: local variable 'curr_task_dir' referenced before assignment

tangxifan, 2021-06-16:
@lukefahr The error is actually means that the script cannot find the task.conf file with a given path to a task directory.

- In most cases, your task should be placed under the ``task`` directory under your OpenFPGA root directory, e.g., https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/tasks
- Alternatively, we also now support running the ``task``  in any location. See details in https://github.com/lnis-uofu/OpenFPGA/issues/209

lukefahr, 2021-06-17:
Hi Xifan, 

I think what you want is for the python to politely say "Task directory X is not found" then die.  What's actually happening is python is crashing on an unbound local variable error because curr_task_dir is not defined. 

-A

tangxifan, 2021-06-17:
@lukefahr You are right. That is I want to improve as well. We will create a PR to solve this issue.

closed 2021-06-18 by ganeshgore