lnis-uofu/OpenFPGA #344: Error when running compilation_verification with blinking.v

Chloe123Jackson, 2021-06-28:
When I run compilation_verification using the blinking.v file i=given in the benchmark, the **00_blinking_MIN_ROUTE_CHAN_WIDTH_out.log**  file is as given below:

RunDirectory : /home/user/OpenFPGA/openfpga_flow/tasks/compilation_verification/run014
python3 /home/user/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/user/OpenFPGA/openfpga_flow/vpr_arch/k4_frac_N4_tileable_adder_chain_mem1K_40nm.xml /home/user/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/blinking/blinking_test/blinking.blif --top_module blinking --run_dir /home/user/OpenFPGA/openfpga_flow/tasks/compilation_verification/run014/k4_frac_N4_tileable_adder_chain_mem1K_40nm/blinking/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/user/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/user/OpenFPGA/openfpga_flow/openfpga_arch/k4_frac_N4_adder_chain_mem1K_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/user/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /home/user/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/blinking/blinking_test/blinking.act --base_verilog /home/user/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/blinking/blinking_test/blinking.v --power --power_tech /home/user/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/user/OpenFPGA/openfpga_flow/tasks/compilation_verification/run014/k4_frac_N4_tileable_adder_chain_mem1K_40nm/blinking/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist
* * * * * * * * * * * * * * * * * * * * 
INFO - Validating commnad line arguments
INFO - Run directory : /home/user/OpenFPGA/openfpga_flow/tasks/compilation_verification/run014/k4_frac_N4_tileable_adder_chain_mem1K_40nm/blinking/MIN_ROUTE_CHAN_WIDTH
INFO - Runing OpenFPGA Shell Engine 
INFO - Launching OpenFPGA Shell Run 
INFO - OpenFPGAShell Revision: 9f58f926-dirty Compiled: 2021-05-07T14:20:51
INFO - OpenFPGA Shell Run is written in file openfpgashell.log
INFO - vpr result extracted in file vpr_stat.result
INFO - Launching iverilog_verification 
ERROR - iverilog_verification run failed with returncode 1
ERROR - command iverilog -o compiled_blinking ./SRC/blinking_include_netlists.v -s blinking_top_formal_verification_random_tb
ERROR - -->>./blinking_output_verilog.v:15: error: out is not a valid l-value in blinking_top_formal_verification_random_tb.REF_DUT.
ERROR - -->>1 error(s) during elaboration.
ERROR - Current working directory : /home/user/OpenFPGA/openfpga_flow/tasks/compilation_verification/run014/k4_frac_N4_tileable_adder_chain_mem1K_40nm/blinking/MIN_ROUTE_CHAN_WIDTH
ERROR - Failed to run iverilog_verification task
ERROR - Exiting . . . . . .

`The blinking_output_verilog.v file is given below:
// ------------------------------
// Design Name: Blinking
// Functionality: 1-bit blinking
// ------------------------------
`timescale 1ns / 1ps
module blinking(
  clk,
  out
);

input clk;
output out;

  always @(posedge clk) begin
    out = ~out;                              //line 15
  end

endmodule`

* * * * * * * * * * * * * * * * * * * * 

I'm not sure of how to resolve this error. It would be helpful if there's a method to do so.

tangxifan, 2021-06-28:
@Chloe123Jackson 

Just tried to modify the task configuration file at the ``compilation_verification`` task. My modification is shown as follows. However, I have not iVerilog reported the errors.

This may be due to the versions of iVerilog. My iverilog version is ``v10.3 (stable)``. 

```
# = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
# Configuration file for running experiments
# = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
# timeout_each_job : FPGA Task script splits fpga flow into multiple jobs
# Each job execute fpga_flow script on combination of architecture & benchmark
# timeout_each_job is timeout for each job
# = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =

[GENERAL]
run_engine=openfpga_shell
power_tech_file = ${PATH:OPENFPGA_PATH}/openfpga_flow/tech/PTM_45nm/45nm.xml
power_analysis = true
spice_output=false
verilog_output=true
timeout_each_job = 20*60
fpga_flow=yosys_vpr

[OpenFPGA_SHELL]
openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/example_script.openfpga
openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml
openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml

[ARCHITECTURES]
arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml

[BENCHMARKS]
bench0=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/blinking/blinking.v

[SYNTHESIS_PARAM]
bench0_top = blinking

[SCRIPT_PARAM_MIN_ROUTE_CHAN_WIDTH]
end_flow_with_test=
vpr_fpga_verilog_formal_verification_top_netlist=
```

Chloe123Jackson, 2021-06-30:
@tangxifan 
Thank you. compilation_verification did work with the modified config file, with verilog 11.0 . May I know why the initial one failed?

tangxifan, 2021-07-01:
@Chloe123Jackson  I guess that previous iVerilog version may require

```
output out; -> output reg out;
```

So that they know the ``out`` is a register.

Chloe123Jackson, 2021-07-05:
@tangxifan  I see. Thank you for the help !

Chloe123Jackson, 2021-07-06:
On running the vcd file, the signal out_out_bench=x. Is that the correct value?

tangxifan, 2021-07-06:
@Chloe123Jackson It is due to Yosys. For some benchmarks, the output is ``x`` due to uninitialized register. However, the FPGA outputs are normally correct. You may try tuning our Yosys script by adding ``write_verilog`` command which should output correct post-synthesis netlists.

Chloe123Jackson, 2021-07-06:
@tangxifan I see. Thank you !  I'll try your suggestion as well

tangxifan, 2022-02-14:
@Chloe123Jackson I have updated yosys https://github.com/lnis-uofu/OpenFPGA/discussions/404
Hope this problem has been solved. If not, feel free to comment and reopen the issue.
Close it for now.

closed 2022-02-14 by tangxifan