lnis-uofu/OpenFPGA #355: OpenFPGA flow fails when an extra reg is added to the and2_latch

abishekshivram, 2021-07-02:
> **Describe the bug**
Hi, I modified and2_latch.v and added a extra reg (attached below) and took it through the OpenFPGA flow using write_full_testbench_example_script and it fails with an error "VprError". I have attached the logs and the task config file below. Let me know how to resolve this. Thank you

[ ] OpenFPGA flow 

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: a2cb153d541d1a2facec65eaa876bd061c6b3748
> 2. Execute OpenFPGA task or your own example: <more details>
> 3. See error

**Verilog benchmark file**:

`timescale 1ns / 1ps

module and2_latch(
  a,
  b,
  clk,
  c,
  d);

input wire clk;

input wire a;
input wire b;
output wire c;
output reg d;
reg temp; // addition from original file

assign c = a & b;

always @(posedge clk) begin
  temp <= c;
  d <= temp;
end

endmodule

**Task Conf file:**

[GENERAL]
run_engine=openfpga_shell
power_tech_file = ${PATH:OPENFPGA_PATH}/openfpga_flow/tech/PTM_45nm/45nm.xml
power_analysis = true
spice_output=false
verilog_output=true
timeout_each_job = 20*60
fpga_flow=yosys_vpr

[OpenFPGA_SHELL]
openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/write_full_testbench_example_script.openfpga
openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k4_N4_40nm_cc_openfpga.xml
openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml
openfpga_vpr_device_layout=
openfpga_fast_configuration=

[ARCHITECTURES]
arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml

[BENCHMARKS]
bench0=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/and2_latch/and2_latch.v

bench1=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/and3/and3.v

bench2=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/or2/or2.v

[SYNTHESIS_PARAM]
bench0_top = and2_latch

bench0_chan_width = 300

bench1_top = and3
bench1_chan_width = 300

bench2_top = or2
bench2_chan_width = 300

[SCRIPT_PARAM_MIN_ROUTE_CHAN_WIDTH]
end_flow_with_test=

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS:  Ubuntu 20.04 

>  - Compiler:

> - Version:
 - [ ] Current master 

> **Additional context**
> Add any other context about the problem here.
[00_and2_latch_MIN_ROUTE_CHAN_WIDTH_out.log](https://github.com/lnis-uofu/OpenFPGA/files/6752150/00_and2_latch_MIN_ROUTE_CHAN_WIDTH_out.log)
[openfpgashell.log](https://github.com/lnis-uofu/OpenFPGA/files/6752151/openfpgashell.log)

abishekshivram, 2021-07-02:
- Using a fixed clock frequency, clock cycles and removing the activity file from the openfpga shell script solves the issue. 
- But if the activity file is included and clock freq is set to auto, this flow fails.

tangxifan, 2021-07-06:
@abishekshivram I looked into the log file. This is due to that VPR removes some redundant nets during optimization which causes the inconsistency error. You may add the option ``--absorb_buffer_luts off`` to disable this net removals.

For example:
https://github.com/lnis-uofu/OpenFPGA/blob/223e06d23c070b4d704a4e120b53e85d738cd162/openfpga_flow/openfpga_shell_scripts/mcnc_example_script.openfpga#L3

abishekshivram, 2021-07-09:
@tangxifan Thanks for the reply. This fixes if the xml file is "k4_N4_40nm_cc_openfpga". However, when we take this modified verilog testbench (and2_latch) through few of frac LUT based xml files like "k4_frac_N4_40nm_cc_openfpga", "k6_frac_N10_adder_chain_40nm_openfpga": the openfpga flow successfully passes, but the verification fails. This fail is that the gfpga signals are at "x" or high.  I have tried using fixed_clock_frequency also. Let me know what you think could be the reason.

abishekshivram, 2021-07-10:
This solves the inconsistency error, thank you @tangxifan

closed 2021-07-10 by abishekshivram