lnis-uofu/OpenFPGA #360: Support Bus Port in Testbench Generator

tangxifan, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
Currently, when there are bus ports in user's HDL designs.
OpenFPGA's testbench generator will flatten these bus ports in testbenches. 
For example, in a counter HDL, there is a 8-bit bus port ``result``
```
module counter (
	clk,
	reset,
	result
);

	input clk;
	input reset;
	output [7:0] result;

	reg [7:0] result;

	always @(posedge clk or posedge reset)
	begin
		if (reset) 
			result = 0;		
		else 
			result = result + 1;
	end
endmodule
```

In the testbench that is auto generated:

```
// ----- Reference Benchmark Instanication -------
	counter REF_DUT(
		clk,
		reset,
		out_result_0__benchmark,
		out_result_1__benchmark,
		out_result_2__benchmark,
		out_result_3__benchmark,
		out_result_4__benchmark,
		out_result_5__benchmark,
		out_result_6__benchmark,
		out_result_7__benchmark	);
```

**Describe the solution you'd like**
The flattened port is due to that VPR can only accept ``.blif`` for now. The ``.blif`` file format does not support bus port. Therefore, we configured yosys to output the flattened netlist.
Once the bus port is flattened, we lost the track how to recover it in downstream flow.

We may consider to extend the [pin constraint file ](https://openfpga.readthedocs.io/en/master/manual/file_formats/pin_constraints_file/) to recover the bus port:

````
<pin_constraints>
  <set_io pin="GPIO[0]" net="result_0_" benchmark_port="result" benchmark_port_index="0"/>
  <set_io pin="GPIO[1]" net="result_1_" benchmark_port="result" benchmark_port_index="1"/>
  ...
</pin_constraints>
```

tornupnegatives, 2022-02-02:
## RTL vs. yosys vs. OpenFPGA
Fortunately, yosys follows a consistent naming convention when flattening array ports: `\name[idx] `. In the below example, snippets from user RTL and yosys synthesis, respectively, are given to demonstrate the translation.

### User RTL
```Verilog
// https://github.com/tornupnegatives/uart/blob/main/rtl/uart_top.v
module uart_top
    (
        // FPGA interface
        input i_clk,
        input i_rst_n,

        // Control interface
        input         i_request_tx,
        input         i_ws_n,
        input         i_rs_n,
        input [3:0]   i_addr,

        // Data busses
        input   [8:0] i_data,
        output  [8:0] o_data,
       // ...
```

### Yosys Synthesis
```verilog
// Generated by Yosys 0.10.0 (git sha1 dca8fb54a, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)
module uart_top
    (
        i_clk,
        i_rst_n,
        i_request_tx,
        i_ws_n,
	i_rs_n,
	\i_addr[0] ,
	\i_addr[1] ,
	\i_addr[2] ,
	\i_addr[3] ,
	\i_data[0] ,
	\i_data[1] ,
	\i_data[2] ,
	\i_data[3] ,
	\i_data[4] ,
	\i_data[5] ,
	\i_data[6] ,
	\i_data[7] ,
	\i_data[8] ,
        // ...
```

With this in mind, it should be possible to edit OpenFPGA's autocheck and formal testbenches such that they instantiate the `REF_DUT` module to follow yosys naming conventions. Below is the `REF_DUT` which OpenFPGA expects, which produces an error, followed by the "correct" version.

### Current OpenFPGA Autocheck
```verilog
uart_top REF_DUT(
	.i_clk(i_clk),
	.i_rst_n(i_rst_n),
	.i_request_tx(i_request_tx),
	.i_ws_n(i_ws_n),
	.i_rs_n(i_rs_n),
	.i_addr_0_(i_addr_0_),
	.i_addr_1_(i_addr_1_),
	.i_addr_2_(i_addr_2_),
	.i_addr_3_(i_addr_3_),
	.i_data_0_(i_data_0_),
	.i_data_1_(i_data_1_),
	.i_data_2_(i_data_2_),
	.i_data_3_(i_data_3_),
	.i_data_4_(i_data_4_),
	.i_data_5_(i_data_5_),
	.i_data_6_(i_data_6_),
	.i_data_7_(i_data_7_),
	.i_data_8_(i_data_8_),
        // ...  
```

### Proposed OpenFPGA Autocheck
```verilog
uart_top REF_DUT(
	.i_clk(i_clk),
	.i_rst_n(i_rst_n),
	.i_request_tx(i_request_tx),
	.i_ws_n(i_ws_n),
	.i_rs_n(i_rs_n),
	.\i_addr[0] (i_addr_0_),
	.\i_addr[1] (i_addr_1_),
	.\i_addr[2] (i_addr_2_),
	.\i_addr[3] (i_addr_3_),
	.\i_data[0] (i_data_0_),
	.\i_data[1] (i_data_1_),
	.\i_data[2] (i_data_2_),
	.\i_data[3] (i_data_3_),
	.\i_data[4] (i_data_4_),
	.\i_data[5] (i_data_5_),
	.\i_data[6] (i_data_6_),
	.\i_data[7] (i_data_7_),
	.\i_data[8] (i_data_8_),
	// ...
```

## Feature Description
OpenFPGA should by default follow yosys' naming convention for array ports. This feature should be implemented such that only the Reference Benchmark Instantiation section of OpenFPGA's testbenches are modified, leaving the testbench declarations themselves alone to preserve compatibility with existing designs.

If the team is OK with this change, I would love to start implementing it!

tangxifan, 2022-02-02:
@tornupnegatives Yes. Let's do this.

tangxifan, 2022-02-15:
@tornupnegatives Let me know if have started the technical work, I am also looking into the issue these days. Trying to work out a solution this week. We can collaborate if you are o.k.

tornupnegatives, 2022-02-15:
@tangxifan Making slow progress as my team is approaching tapeout, but I am planning to dedicate more time to it this week. I am happy to collaborate. Feel free to email me!

tangxifan, 2022-02-15:
@tornupnegatives No worries. As we discussed before, there are several ways to solve this issue:

- OpenFPGA has an internal rule when folding the pins. For example, we can follow the rule that yosys is doing, as you showed
```
.i_addr(.\i_addr[0], .\i_addr[1] ... )
```

- OpenFPGA can accept an external file (e.g., an XML file), which contains the detailed rules. For example,
```
<bus_groups>
  <bus name="i_addr" lsb="0" msb="3">
    <pin id="0" name="i_addr_0_"/>
    <pin id="1" name="i_addr_1_"/>
    <pin id="2" name="i_addr_2_"/>
    <pin id="3" name="i_addr_3_"/>
  </bus>
</bus_groups>
```

I believe the first way is a low hanging fruit while the second way takes more time. But we never know after trying. 
I will try locally and see if what you have proposed works or not.
I will also read the source codes again and keep you posted on how to modify.

tangxifan, 2022-02-18:
@tornupnegatives Once PR #539 is merged, OpenFPGA support an external file which can recover buses in verilog testbenches.