lnis-uofu/OpenFPGA #361: Support Fully Customizable Scan-chain in Fabric Netlist Generator

tangxifan, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
Scan chain has been widely used in SOFA project because enables testable FPGA fabric.
The scan chain organization follows a snake-like organization.

[image]

However, the scan-chain insertion is not automated yet in OpenFPGA's fabric generator. A lot of manual work is required.
Currently, the scan-chain connection support is very limited, as shown in 

https://openfpga.readthedocs.io/en/master/manual/arch_lang/direct_interconnect/#example

[image]

Such connection will cause long interconnecting wires in scan-chain, slowing down the operating speed and inducing large area overhead.

Plus the head and tail of scan-chain is not automatically generated and wired in OpenFPGA's fabric generator.

So the scan-chain support has to be significantly reworked.

**Describe the solution you'd like**

- We should deprecate the https://openfpga.readthedocs.io/en/master/manual/arch_lang/direct_interconnect/#directlist
- Then consider a scan-chain key file, similar to the [fabric key file](https://openfpga.readthedocs.io/en/master/manual/file_formats/fabric_key/) 
As such, we can enable fully customizable scan-chain generation.
- Also, the scan-chain head and tail should be identified through new syntax in OpenFPGA architecture description file.
As a result, the head and tail ports can be automatically generated in the FPGA netlists.

For example,

```
<scanchain>
  <head tile="clb" x="6" y="6" port="sc_in"/>
  <tail tile="clb" x="6" y="1" port="sc_out"/>
   <key file="scan_chain_connection.xml"/>
</scanchain> 
```

ganeshgore, 2021-07-13:
Why deprecate the `Directlist` its useful in general right ?

tangxifan, 2021-07-13:
@ganeshgore It is barely used actually in our tape-outs.