lnis-uofu/OpenFPGA #365: Accept absolute project paths in `run_fpga_task.py` script?

WRansohoff, 2021-08-13:
**Is your feature request related to a problem? Please describe.**
It would be nice to be able to run OpenFPGA tasks using an absolute path to specify the task location. For example:

`python3 $OPENFPGA_PATH/openfpga_flow/scripts/run_fpga_task.py /home/[user]/my_fpga_task`
`run-task /home/[user]/my_fpga_task`

**Describe the solution you'd like**
Currently, the above calls will fail even when a valid OpenFPGA task directory is located at `/home/[user]/my_fpga_task`. It looks like `run_fpga_task.py` only looks for tasks in a couple of local subdirectories.

**Describe alternatives you've considered**
Moving every OpenFPGA task into the repository's `openfpga_flow/tasks/` directory seems a bit cumbersome, especially if the tasks might be located on another drive or filesystem.

**Additional context**
[Here's one potential solution.](https://github.com/WRansohoff/OpenFPGA/commit/c31c1d8b049e1cc09fdcb2850d564971c51b0a44)

ganeshgore, 2021-08-13:
Sounds good. Would you please make a PR?

WRansohoff, 2021-08-13:
Sure: #366. Thank you!

tangxifan, 2021-08-13:
Just a reminder, tests related to the new feature should be added to the CI.

tangxifan, 2021-08-13:
@WRansohoff I have discussed with @ganeshgore about the PR, the following requirements are need to guarantee the quality of the feature:
- A simple test should be added to the CI, as a validation on an working example on the new feature
- Documentation should be updated to guide users about using the new feature
- The added python codes requires some comment to help coders to under it easily (May not required if the code is straightforward).

Please let me know if you are interested in creating a few follow-up pull requests on this effort.
If you see any issue, feel free to contact me.

WRansohoff, 2021-08-15:
Sure thing, thank you for the feedback. I will submit a follow-up PR shortly with a CI test, docs update, and comment.

tangxifan, 2021-08-16:
> Sure thing, thank you for the feedback. I will submit a follow-up PR shortly with a CI test, docs update, and comment.

@WRansohoff  Thanks for the contribution. Feel free to contact me if you need any help.

tangxifan, 2022-06-02:
Already covered in regression tests: 

https://github.com/lnis-uofu/OpenFPGA/blob/70e2330f1740786ac7a108fd4182d63ab75069d3/openfpga_flow/regression_test_scripts/basic_reg_test.sh#L195-L199

closed 2022-06-02 by tangxifan