lnis-uofu/OpenFPGA #373: Possible partial data problem in vpr_stat.result

foggy-slt, 2021-09-17:
Hi

We found a possibile bug when extracting the data.

The reproducing steps are shown as in the following.

1) Add the EPFL benchmark suits under dir **OpenFPGA/openfpga_flow/benchmarks**

2) Add the **yosys.ys** script under dir **OpenFPGA/openfpga_flow/misc**

```bash
${READ_VERILOG_FILE}
synth
write_blif ${OUTPUT_BLIF}
```

3) run the flow

4) Check the file **task_result.csv**, the columns **total_routing_area** and **total_logic_block_area** seems missing information.

[image]

We modified the regular expressions in file **fpgaflow_default_tool_path.conf** of  dir **OpenFPGA/openfpga_flow/misc**.

```perl
total_routing_area = "Total routing area: ([0-9.]+e\+[0-9.]+|[0-9.]+)", str 
total_logic_block_area = "Total used logic block area: ([0-9.]+e\+[0-9.]+|[0-9.]+)", str
```

Now it works, please double check.

Thanks.

tangxifan, 2021-09-17:
@ganeshgore I doubled checked the conf file. The proposed changes make sense to me. I do not know if the ``.`` can cover any A-Z or a-z codes. 

Can you confirm if this is really an issue?

https://github.com/lnis-uofu/OpenFPGA/blob/30feb78469dc6132c0d66485ba4e91254b0de1a6/openfpga_flow/misc/fpgaflow_default_tool_path.conf#L27

tangxifan, 2021-09-17:
@foggy-slt Just a small tweak, we need to consider both cases: ``1e9`` and ``1E9`` in regular expression.

ganeshgore, 2021-09-17:
@foggy-slt Thank you for your note, I guess I did not consider the scientific notations while writing that regex. We should update that. The general format of the VPR log is as follows 

```
	Total logic block area (Warning, need to add pitch of routing to blocks with height > 3): 3.33e+06
	Total used logic block area: 2.28e+06

Routing area (in minimum width transistor areas)...
	Total routing area: 481319., per logic tile: 2848.04
```

I am not sure why you added additional `|[0-9.]+`. Please let me know if I am missing something again.
But  We can go with the following regex
`"Total routing area: ([0-9.]+[e|E\+[0-9]+)", str `

If this works for you, please make a pull request with all `([0-9.]+)` regex updated with the above version.

closed 2021-09-27 by ganeshgore