lnis-uofu/OpenFPGA #377: Using 22nm Architectures

Iripi97, 2021-09-22:
I am trying to use a 22nm architecture/feature size for my OpenFPGA project; however I am unable to find the 22nm architecture files (.XML). I can easily find the files required for Power analysis (in OpenFPGA/openfpga_flow/tech/PTM_22nm) with the 22nm feature size, however I cannot find the architecture files for 22nm in either /OpenFpga/openfpga_flow/openfpga_arch or in /OpenFPGA/openfpga_flow/vpr_arch.

Are there any 22nm sample .XML files (both for the openfpga_arch AND vpr_arch directories) that could be added to the repository, or linked in a comment below?

tangxifan, 2021-09-23:
@Iripi97 Currently, we do not have any example architecture using a 22nm technology. However, OpenFPGA offers you high flexibility when porting an architecture from a technology to another.  Please checkout the tutorial https://openfpga.readthedocs.io/en/master/tutorials/arch_modeling/open_cell_libraries_tutorial/, with which you can adapot existing examples to a 22nm technology.

Iripi97, 2021-09-23:
@tangxifan I have already seen this tutorial. It is extremely helpful for understanding when to use different cell libraries. However at the moment, I do not have access to 22nm XML files, so realistically the main thing I am trying to change in files like _k4_N4_40nm.xml_ or _k4_N4_tileable_40nm.xml_ is the Vdd, chan_length, min_width, etc... essentially values that relate to the objects sizes/power that are declared in the .xml files. Realistically I am fine with just using the standard cell library, I just want to make sure the fabric is being accurately modeled size/power wise for 22nm. Also does one just need to modify the task.config file for the task to ensure PPA numbers are generated? Also where are these numbers generated in the contexts of the OpenFPGA workspace? Is it within the directory that houses the current task's related files/configurations?

tangxifan, 2021-09-27:
@Iripi97

> I do not have access to 22nm XML files, so realistically the main thing I am trying to change in files like k4_N4_40nm.xml or k4_N4_tileable_40nm.xml is the Vdd, chan_length, min_width, etc... essentially values that relate to the objects sizes/power that are declared in the .xml files.

-  The ``Vdd``, ``chan_length``, ``min_width`` are parameters required by FPGA-SPICE. If you do not care FPGA-SPICE, you can give dummy numbers, FPGA-Verilog and FPGA-Bitstream should work fine.

>  Realistically I am fine with just using the standard cell library, I just want to make sure the fabric is being accurately modeled size/power wise for 22nm. Also does one just need to modify the task.config file for the task to ensure PPA numbers are generated?  Realistically I am fine with just using the standard cell library, I just want to make sure the fabric is being accurately modeled size/power wise for 22nm. Also does one just need to modify the task.config file for the task to ensure PPA numbers are generated? 

- In general, it is chip designers' responsibility to guarantee the correct PPA numbers in FPGA architecture description. It means that you need to run some timing analysis using ASIC tools, e.g., Synopsys PrimeTime or Synopsys HSPICE, and then you annotate the timing values to VPR architecture XML. As a result, your FPGA architecture reflects the correct timing of your chip.

You may consider to post such question on the discussion forum. Since this is more like a topic to discuss.

Iripi97, 2021-09-28:
@tangxifan Thank you for your responses, they are helpful. Just one more question that I would love some clarification on:
In my task.conf file I have the parameters _power_analysis_, _spice_output_, _verilog_output_ all set to true (as well as _power_tech_file_ pointing to appropriate XML file) but I am failing to generate (or see where it is generated) Power results. I can see the timing and area results in vpr_stat.out. Is this where Power results should be found, or is it typically generated in another file?

tangxifan, 2021-10-05:
@Iripi97 Power analysis of VPR is enabled by adding options ``--power`` to your openfpga shell script when calling ``vpr``. Currently, there is no such example in OpenFPGA's repo now. But it is very easy to do. 
- You can check the VTR documentation at https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/?highlight=power#power-estimation-options
- Update your openfpga shell script by adding the options to the line where we call VPR. For example: 
https://github.com/lnis-uofu/OpenFPGA/blob/1603c9b40413a4256c7718b72d0cb40183bfb88b/openfpga_flow/openfpga_shell_scripts/mcnc_example_script.openfpga#L3
- Run the openfpga flow, the power results are stored in a separated file under postfix ``.power``

tangxifan, 2021-12-10:
@Iripi97 For power analysis, you need
- enable ``--power`` option
- provide a tech property file (.xml)
- provide a signal activity file (.act)

Please read the VTR documentation https://docs.verilogtorouting.org/en/latest/vtr/power_estimation/#vpr

tangxifan, 2022-09-06:
Close the issue as answered. If you need further help, please comment (reopen this issue) or create a new issue.
Thank you for the interests in OpenFPGA.

closed 2022-09-06 by tangxifan