lnis-uofu/OpenFPGA #384: Error in Netlist Generation  **Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH**

Mistal1999, 2021-10-05:
INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task basic_tests/full_testbench/configuration_chain
 INFO (     MainThread) - Created "run006" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 1 Architectures 3 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 3 jobs
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - **Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH**
Traceback (most recent call last):
  File "/data/lizeju/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 476, in run_single_script
    raise subprocess.CalledProcessError(0, " ".join(command))
subprocess.CalledProcessError: Command 'python3 /data/lizeju/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /data/lizeju/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /data/lizeju/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --top_module and2 --run_dir /data/lizeju/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run006/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /data/lizeju/OpenFPGA/openfpga_flow/openfpga_shell_scripts/write_full_testbench_example_script.openfpga --openfpga_arch_file /data/lizeju/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_cc_openfpga.xml --openfpga_sim_setting_file /data/lizeju/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --openfpga_vpr_device_layout  --openfpga_fast_configuration  --power --power_tech /data/lizeju/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /data/lizeju/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run006/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --TOP and2 --CHAN_WIDTH 300' returned non-zero exit status 0.

 I have installed all the requirments but it still doesn't work.

tangxifan, 2021-10-05:
@Mistal1999  Can you share your openfpgashell.log file?

Mistal1999, 2021-10-06:
I know is a little bit embrass. But I can't locate  the openfpgashell.log file. Could you please tell the path of it? Thanks a lot

Mistal1999, 2021-10-06:
[openfpgashell.log](https://github.com/lnis-uofu/OpenFPGA/files/7292838/openfpgashell.log)
Is it this file?

closed 2021-10-16 by Mistal1999