lnis-uofu/OpenFPGA #395: port has already been declared when complied by iverilog

Mistal1999, 2021-10-20:
iverilog -o compiled_counter120bitx5 ./SRC/counter120bitx5_include_netlists.v -s counter120bitx5_autocheck_top_tb

lizeju@amax:/data/lizeju/OpenFPGA/openfpga_flow/tasks/quicklogic_tests/counter_5clock_test/run008/k4_N4_tileable_GlobalTile8Clk_40nm/counter120bitx5/MIN_ROUTE_CHAN_WIDTH$ iverilog -o compiled_counter120bitx5 ./SRC/counter120bitx5_include_netlists.v -s counter120bitx5_autocheck_top_tb
**_././SRC/counter120bitx5_autocheck_top_tb.v:71: error: 'reset' has already been declared in this scope.
././SRC/counter120bitx5_autocheck_top_tb.v:18:      : It was declared here as a net._**

Many auto generated tb files has this common error when they are complied by iverilog, such as fifo
How do solve this problem.

tangxifan, 2021-10-29:
@Mistal1999 Sorry a late reply. This is indeed a bug. Currently, the ``reset`` is a reserved signal in openfpga's testbench generator. Can you modify the testbench file to avoid the problem?

msaideroglu, 2021-11-09:
I'm facing with same issue

msaideroglu, 2021-11-09:
When I try to generate a counter testbench with generate_testbench task, it gives me same errors:
The configuration and log files attached
[config.txt](https://github.com/lnis-uofu/OpenFPGA/files/7507329/config.txt)
:
[log.txt](https://github.com/lnis-uofu/OpenFPGA/files/7507331/log.txt)

Mistal1999, 2021-11-13:
Thank you for your responses, they are helpful.

tangxifan, 2022-02-14:
@msaideroglu Sorry for late reply. The problem comes from the ``reset`` net is both a pin of FPGA global ports and a input of the reference benchmark. As a result, openfpga added two ports (wire/register) to the full testbench

````
// Global port
wire reset;
// Shared inputs
reg reset;
```

The above syntax is o.k. to icarus Verilog but complained by commercial tools, such as Modelsim.
I have created PR #528  
It should address the issue.

closed 2022-02-14 by tangxifan