lnis-uofu/OpenFPGA #405: QuickLogic test is currently off due to Yosys v0.10 upgrade

tangxifan, 2021-10-31:
> **Describe the bug**
> A clear and concise description of what the bug is.

- [x] OpenFPGA flow

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: 4e50644
> 2.  QuickLogic tests are disabled because their synthesis recipe is not yet in the Yosys v0.10 release

https://github.com/lnis-uofu/OpenFPGA/blob/4e50644ea866e30ea8dadb67d127a27f8fa6375a/openfpga_flow/regression_test_scripts/quicklogic_reg_test.sh#L11-L21

> **Expected behavior**
> Should enable it soon so that the soft adder support is tested.

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

tangxifan, 2021-10-31:
@tpagarani @coolbreeze413 You may consider to bring the Quicklogic scripts to the upstream and we can bump up the yosys submodule.

coolbreeze413, 2021-11-18:
Added tests with #420

tangxifan, 2021-11-18:
@coolbreeze413 Just a reminder. There are still a few quicklogic tests off: counter_5clock_test, sdc_controller_test, lut_adder_test.
When they are all back. We can safely close this issue

coolbreeze413, 2021-11-18:
@tangxifan I will check about adding those as well

tangxifan, 2021-12-10:
All done by PR #420 and #421

closed 2021-12-10 by tangxifan