lnis-uofu/OpenFPGA #406: Missing cairo.h when trying to build OpenFPGA with GUI enabled

MidsummerNight, 2021-11-02:
Sorry if this turns out to be a low level mistake on my part, but when following the [How to Compile](https://openfpga.readthedocs.io/en/latest/tutorials/getting_started/compile/) instructions on enabling the GUI ("VPR’s GUI requires gtk-3, and can be enabled with  ```cmake .. -DVPR_USE_EZGL=on```"), I modified /OpenFPGA/Makefile like this:

```
# Makefile

ifeq ($(origin CMAKE_COMMAND),undefined)
CMAKE_COMMAND := cmake
else
CMAKE_COMMAND := ${CMAKE_COMMAND}
endif

.PHONY: all checkout compile

all: checkout
	mkdir -p build && cd build && $(CMAKE_COMMAND) ${CMAKE_FLAGS} .. -DVPR_USE_EZGL=on
	cd build && $(MAKE)

checkout: 
	git submodule init
	git submodule update --init --recursive

compile:
	mkdir -p build && cd build && $(CMAKE_COMMAND) ${CMAKE_FLAGS} .. -DVPR_USE_EZGL=on
	cd build && $(MAKE)

clean:
	rm -rf build

build/Makefile:
	make checkout

.PHONY: Makefile

%: build/Makefile
	cd build && $(MAKE) $@
```
Then executed ```make all```, and everything went well until this:
```
......<left out previous output results>......
make[3]: Leaving directory '/home/my_machine/OpenFPGA/build'
[  3%] Built target libtatum
make[3]: Entering directory '/home/my_machine/OpenFPGA/build'
Scanning dependencies of target ezgl
make[3]: Leaving directory '/home/my_machine/OpenFPGA/build'
make[3]: Entering directory '/home/my_machine/OpenFPGA/build'
[  3%] Building CXX object libs/EXTERNAL/libezgl/CMakeFiles/ezgl.dir/src/application.cpp.o
In file included from /home/my_machine/OpenFPGA/libs/EXTERNAL/libezgl/include/ezgl/canvas.hpp:24,
                 from /home/my_machine/OpenFPGA/libs/EXTERNAL/libezgl/include/ezgl/application.hpp:22,
                 from /home/my_machine/OpenFPGA/libs/EXTERNAL/libezgl/src/application.cpp:19:
/home/my_machine/OpenFPGA/libs/EXTERNAL/libezgl/include/ezgl/graphics.hpp:27:10: fatal error: cairo.h: No such file or directory
   27 | #include <cairo.h>
      |          ^~~~~~~~~
compilation terminated.
make[3]: *** [libs/EXTERNAL/libezgl/CMakeFiles/ezgl.dir/build.make:63: libs/EXTERNAL/libezgl/CMakeFiles/ezgl.dir/src/application.cpp.o] Error 1
make[3]: Leaving directory '/home/my_machine/OpenFPGA/build'
make[2]: *** [CMakeFiles/Makefile2:980: libs/EXTERNAL/libezgl/CMakeFiles/ezgl.dir/all] Error 2
make[2]: Leaving directory '/home/my_machine/OpenFPGA/build'
make[1]: *** [Makefile:141: all] Error 2
make[1]: Leaving directory '/home/my_machine/OpenFPGA/build'
make: *** [Makefile:13: all] Error 2
```

To be honest I didn't really understand how to ```cmake .. -DVPR_USE_EZGL=on``` exactly, so I took a guess and added the flag inside the Makefile. Is this not the proper way and the reason of my error? Or did I mishandle something else? Many thanks!

tangxifan, 2021-11-02:
@MidsummerNight 

From the log file, I think you missed a package:

https://github.com/lnis-uofu/OpenFPGA/blob/4e50644ea866e30ea8dadb67d127a27f8fa6375a/.github/workflows/install_dependencies_build.sh#L17

MidsummerNight, 2021-11-02:
@tangxifan Yeah I noticed that right after posting, sorry my fault. But is adding  ```-DVPR_USE_EZGL=on``` directly inside the  Makefile the proper way to enable the GUI?

MidsummerNight, 2021-11-02:
Update:

The same issue persists even after installing the packages listed in [OpenFPGA/.github/workflows/install_dependencies_build.sh](https://github.com/lnis-uofu/OpenFPGA/blob/4e50644ea866e30ea8dadb67d127a27f8fa6375a/.github/workflows/install_dependencies_build.sh). Strangely `-- EZGL: graphics enabled` is always displayed before `cd build && make`, regardless of whether `-DVPR_USE_EZGL=on` is in Makefile or not, and the build process succeeded after I installed ```libgtk-3-dev```.

tangxifan, 2021-11-02:
@MidsummerNight In general, you do not need to specify the ``-DVPR_USE_EZGL=on`` using Cmake options. The Cmakefile is designed to automatically detect your packages, if the packages are all installed, if will enable GUI. Otherwise it will not.

MidsummerNight, 2021-11-03:
In the end, my issue was solved by installing ```libgtk-3-dev```, whatever the reason. This is indeed quite strange since I recall  successfully building OpenFPGA back in September without `libgtk-3-dev`. I'll close this issue now, hopefully my experience would be helpful to future troubled users.

closed 2021-11-03 by MidsummerNight