lnis-uofu/OpenFPGA #407: Quick Compilation Verification fails

Sun1927, 2021-11-03:
After compile OpenFPGA as what is shown in the website, I am trying to verify if the tool is well compiled by run the command "python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification", there is an error to execute openfpga flow. This is the running result:

sun@sun-virtual-machine:~/OpenFPGA$ python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification
 INFO (     MainThread) - Created "run002" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
ERROR (00_blinking_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_blinking_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 479, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/sun/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/sun/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/blinking/blinking.v --top_module blinking --run_dir /home/sun/OpenFPGA/openfpga_flow/tasks/compilation_verification/run002/k4_N4_tileable_40nm/blinking/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/sun/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/sun/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/sun/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --power --power_tech /home/sun/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/sun/OpenFPGA/openfpga_flow/tasks/compilation_verification/run002/k4_N4_tileable_40nm/blinking/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist --TOP blinking' returned non-zero exit status 0.

It will be grateful for me if you can give some help. Thanks for your reading.

ganeshgore, 2021-11-03:
Can you run the task with the following flags and share the log file?

https://github.com/lnis-uofu/OpenFPGA/blob/1603c9b40413a4256c7718b72d0cb40183bfb88b/openfpga_flow/regression_test_scripts/basic_reg_test.sh#L12

Sun1927, 2021-11-04:
Thanks for your reply.

Sorry I am not familiar with Linux system and command, but do you mean this?

I tried to run the task again, with "python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/full_testbench/configuration_chain", got the same result.

 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task basic_tests/full_testbench/configuration_chain
 INFO (     MainThread) - Created "run004" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 1 Architectures 3 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 3 jobs
ERROR (00_or2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_or2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 479, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/sun/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/sun/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/or2/or2.v --top_module or2 --run_dir /home/sun/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run004/k4_N4_tileable_40nm/or2/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/sun/OpenFPGA/openfpga_flow/openfpga_shell_scripts/write_full_testbench_example_script.openfpga --openfpga_arch_file /home/sun/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_cc_openfpga.xml --openfpga_sim_setting_file /home/sun/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --openfpga_vpr_device_layout  --openfpga_fast_configuration  --power --power_tech /home/sun/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/sun/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run004/k4_N4_tileable_40nm/or2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --TOP or2 --CHAN_WIDTH 300' returned non-zero exit status 0.

tangxifan, 2021-11-04:
@Sun1927 What @ganeshgore suggests is to add ``--debug`` and ``--show_thread_logs`` options so that you will see detailed error messages.

Sun1927, 2021-11-05:
@tangxifan thanks for your help. The problem has been solved by install envyaml.

closed 2021-11-08 by tangxifan